RobotWebTools/roslibjs #152: New TF implementation not working

rctoris, 2015-02-04:
@T045T I released the new version of roslibjs with the new TF implementation and TFs are no longer working correctly. Looks like no TFs are handled correctly (see screenshot). 

[image]

Since this is a critical bug I'm going to revert and re-release!

T045T, 2015-03-18:
Can you give a minimal (non-)working example? I've tested the new TF implementation a number of times and never run into this.

rctoris, 2015-03-18:
The example above is from the RWT demo https://github.com/RobotWebTools/robotwebtools.github.io/blob/master/demos/carl-urdf.html 

It's running on Indigo with:

`sudo apt-get install ros-indigo-carl-description`

```
<launch> 
  <include file="$(find rosbridge_server)/launch/rosbridge_websocket.launch" />
  <node pkg="tf2_web_republisher" type="tf2_web_republisher" name="tf2_web_republisher" />
  <include file="$(find carl_description)/launch/carl_description.launch" />
</launch>
```

PR #171 (2015-03-27): Restore tf2_web_republisher service functionality
closes #152 - turns out ```trimLeft```was the culprit
closed 2015-03-27 by rctoris