RobotWebTools/roslibjs #155: Websocket should use BSON to properly support PointClouds and other binary data

pbeeson, 2015-02-06:
I was using the rosbridge websocket server for a project, but we were sending LARGE amounts of point cloud data to a web page to be rendered.  It turns out of course that there were several issues.  The first of which is the fact that the roslibjs client side does keep up and memory leaks occur.  This was easily fixed by setting up a service to request large PC data via pulling instead of attaching to a broadcasting stream of hundread of MB of data a second.

However, we quickly noticed that in roslibjs, the majority of the time taken to render the PoinCloud was in decoding the base64 encoding of the binary data.  After looking into this, it turns out that with VERY minimal changes, rosbridge and roslibjs can be set up to use BSON (just 'apt-get install python-bson' and get the bson.js) and point clouds can be passed as binary blobs with no problem.  We've been running this for weeks and see a speedup from 65ms to unpack and render a 3D point cloud from a Carnegie Robotics Multisense to a 15ms.  That's a HUGE performance increase.

Below I tack on the siff to show how simplt my changes are, and would like to encourage rosbridge and roslibjs devels to switch to using BSON for server to client message passing, while still using JSON for client to server message, service, param comms as that is typically very small amounts of data.

```
pbeeson@tardis:~/projects/PROWL/code/rosworkspace$ diff -r rosbridge_suite/src ~/Desktop/rosbridge_suite-develop | grep -v pyc
Only in rosbridge_suite/src: CMakeLists.txt
diff -r rosbridge_suite/src/rosbridge_library/src/rosbridge_library/internal/message_conversion.py /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_library/src/rosbridge_library/internal/message_conversion.py
43d42
< import bson
104c103
<             return bson.Binary(inst)
---
>             return standard_b64encode(inst)
diff -r rosbridge_suite/src/rosbridge_library/src/rosbridge_library/protocol.py /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_library/src/rosbridge_library/protocol.py
42d41
< import bson
264,265c263
<             #return json.dumps(msg)
<             return bson.BSON.encode(msg)
---
>             return json.dumps(msg)
diff -r rosbridge_suite/src/rosbridge_server/scripts/rosbridge_websocket /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_server/scripts/rosbridge_websocket
105c105
<         IOLoop.instance().add_callback(partial(self.write_message, message, True))
---
>         IOLoop.instance().add_callback(partial(self.write_message, message))
diff -r rosbridge_suite/src/rosbridge_server/scripts/rosbridge_websocket.py /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_server/scripts/rosbridge_websocket.py
105c105
<         IOLoop.instance().add_callback(partial(self.write_message, message, True))
---
>         IOLoop.instance().add_callback(partial(self.write_message, message))
```

And for roslib.js
```
pbeeson@tardis:~/projects/PROWL/code/BrowserUI/modified$ diff roslib.js ~/Desktop/roslib.js 
1,2d0
< var BSON = bson().BSON;
< 
1092,1102c1090,1095
< 	var reader    = new FileReader();
< 	reader.onload  = function() {
< 	    var uint8Array = new Uint8Array(this.result);
< 	    var msg = BSON.deserialize(uint8Array);
<   	    if (msg.op === 'png') {
< 		decompressPng(msg, handleMessage);
< 	    } else {
< 		handleMessage(msg);
< 	    }
< 	}
< 	reader.readAsArrayBuffer(message.data);
---
>       var data = JSON.parse(typeof message === 'string' ? message : message.data);
>       if (data.op === 'png') {
>         decompressPng(data, handleMessage);
>       } else {
>         handleMessage(data);
>       }
```

pbeeson, 2015-02-06:
Also added this as an issue in rosbridge_suite as it affects both server and client.
https://github.com/RobotWebTools/rosbridge_suite/issues/169

rctoris, 2015-02-06:
Closing to keep discussion in a single forum (https://github.com/RobotWebTools/rosbridge_suite/issues/169). This is a major change to the rosbridge protocol itself so it makes sense to keep discussion there as it would effect multiple client implementations, not just JavaScript.

closed 2015-02-06 by rctoris

naveedhd, 2015-05-12:
@pbeeson how would you include bson package in roslibjs?