RobotWebTools/roslibjs #157: Separate URDF functionality

jakobs, 2015-02-13:
I would like to use the urdf js code for some work that is unrelated to ROS. Do you think it would be possible to separate the ROS specific parts from the URDF parts? I had a quick look, and it seems to be not too much work. 
The urdf stuff could go into a new repo, which would also include all the three.js adapters in the visualisation library.

megawac, 2015-02-14:
@jakobs checkout #158. If you use a build tool such as browserify, webpack or esperanto you can omit `urdf` in your build by requirign components directly

(if #158 is merged)
```js
var Ros = require('roslib/src/core');
```

jakobs, 2015-02-14:
Well, I wanted only the urdf stuff. Currently I used browserify to extract it, but this is a manual thing. I would have liked to have urdf both from rslibjs and from ros3djs in a separate repo, and in its own namespace, so I can include it in my build chain easily. In general I would think RobotWebTools would benefit by separating the ROS related things.

megawac, 2015-02-14:
In that case

```js
var urdf = require('roslib/src/urdf');
```

I dig how this package is the base of what you'll likely need

rctoris, 2015-02-15:
The problem with making the URDF component of roslib and ros3d its own package is then making it a requirement for both roslib and ros3d later on. Combining the two and making it a dep for roslib then means you have 3d visualization code floating around in an app that doesn't need it. Also, this would break everyone's code if this standalone lib was now not included in roslibjs.

jakobs, 2015-02-15:
Thanks @rctoris for the clarification. I do understand your point. The hope I had was that you could invert the dependency chain. Generic JS code, which is used to visualize robot interfaces, and a ros specific library for specialisation. But I appreciate that this would be significant work, and there is no real benefit for someone using ROS.

PR #158 (2015-02-15): Index components of roslib
This allows users to do

```js
> var Ros = require("roslib/src/core")

{ Ros: [Function: Ros],
  Topic: [Function: Topic],
  Message: [Function: Message],
  Param: [Function: Param],
  Service: [Function: Service],
  ServiceRequest: [Function: ServiceRequest],
  ServiceResponse: [Function: ServiceResponse]
}
```

Fixes #157
closed 2015-02-15 by jakobs