RobotWebTools/roslibjs #160: node package fails to "build" with node v0.12.0

chrisl8, 2015-02-21:
The install is failing in node v0.12.0. It works with the older 0.10.35.
The failure appears to be related to the dependency on canvas 1.1.3

```
5520 verbose about to build /home/chrisl8/node_modules/roslib/node_modules/xmlshim
5521 verbose unlock done using /home/chrisl8/.npm/_locks/xmlshim-c8d70b8da5989174.lock for /home/chrisl8/node_modules/roslib/node_modules/xmlshim
5522 verbose about to build /home/chrisl8/node_modules/roslib
5523 verbose unlock done using /home/chrisl8/.npm/_locks/roslib-8c9bde6a3240ef46.lock for /home/chrisl8/node_modules/roslib
5524 verbose stack Error: canvas@1.1.3 install: `node-gyp rebuild`
5524 verbose stack Exit status 1
5524 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/lifecycle.js:213:16)
5524 verbose stack     at EventEmitter.emit (events.js:110:17)
5524 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/spawn.js:14:12)
5524 verbose stack     at ChildProcess.emit (events.js:110:17)
5524 verbose stack     at maybeClose (child_process.js:1008:16)
5524 verbose stack     at Process.ChildProcess._handle.onexit (child_process.js:1080:5)
5525 verbose pkgid canvas@1.1.3
5526 verbose cwd /home/chrisl8
5527 error Linux 3.13.0-45-generic
5528 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "roslib"
5529 error node v0.12.0
5530 error npm  v2.5.1
5531 error code ELIFECYCLE
5532 error canvas@1.1.3 install: `node-gyp rebuild`
5532 error Exit status 1
5533 error Failed at the canvas@1.1.3 install script 'node-gyp rebuild'.
5533 error This is most likely a problem with the canvas package,
5533 error not with npm itself.
5533 error Tell the author that this fails on your system:
5533 error     node-gyp rebuild
5533 error You can get their info via:
5533 error     npm owner ls canvas
5533 error There is likely additional logging output above.
5534 verbose exit [ 1, true ]
5535 verbose unbuild node_modules/roslib/node_modules/canvas
5536 info preuninstall canvas@1.1.3
5537 info uninstall canvas@1.1.3
```

jihoonl, 2015-03-10:
Thank you for the report. Will catch this up later.

T045T, 2015-04-26:
Sadly, this seems to be a known (and thus far, unadressed) problem with node-canvas: https://github.com/Automattic/node-canvas/issues/536

chrisl8, 2015-07-01:
If I make these two changes to package.json then "npm install" works:

First, I think the required version of canvas is just too old:
-    "canvas": "1.1.3",
+    "canvas": ">=1.1.3",

Second, the existing xmlshim does not work on node 0.12. There are forks that do, but the pull requests have not been merged, so pointing to galvanix's "greenfelt" branch allows it to build:
-    "xmlshim": "~0.0.9"
+    "xmlshim": "https://github.com/galvanix/node-xmlshim/tarball/greenfelt"

However I haven't fully tested if all of this works. So I'm not making any pull requests yet.
Just posting my findings here.

michaelryancaputo, 2015-08-20:
Package is unusable in it's current state. Any plans to fix?

jihoonl, 2015-09-02:
Fixed in 0.17.0
https://github.com/RobotWebTools/roslibjs/pull/193

closed 2015-09-02 by jihoonl