RobotWebTools/roslibjs #162: Serive calls are slow

pbeeson, 2015-02-23:
I've noticed that service calls are generally AT LEAST 100 ms from the roslibjs call to when the callback is called, even for empty messages and a service that does no work (just returns True).

What is going on that makes service calls through roslib so slow?

pbeeson, 2015-02-24:
I think this boils down to the roslib.js client not pulling data off the websocket fast enough, as tornado on the rosbridge_server is getting many WOULDBLOCK errors on the socket send.   Maybe I'm missing a parameter to the WebSocket stuff that would improve throughput?  In an case, I think this may be lower level than roslib.js.

closed 2015-02-24 by pbeeson

smamessier, 2015-02-25:
Check issue #171 on Rosbridge_suite, it might be your problem too. I'm able to go way below 30ms by using TCP_nodelay on the server side.

https://github.com/RobotWebTools/rosbridge_suite/issues/171