RobotWebTools/roslibjs #165: Source comment for topic unsubscribe is unclear

pbeeson, 2015-02-27:
The comment in roslib.js for Topic.prototype.unsubscribe states "Unsubscribing will remove all subscribe callbacks.".   This is not true.  Unsubscribing will turn off all callbacks, but does not remove them.  So that when resubscribing, callbacks will start back up again.  The only way to remove callbacks is to call unsubscribe with the callback as an argument. 

This really affected me because I was often unsubscribing/subscribing in my Browser program and each unsubscribe had no arguments while each subscribe call was sending the same callback.  What happened was that each subscribe call would make ANOTHER callback, so that after calling subscribe N times, I had N of the same callbacks for each incoming message.  I had assumed that unsubscribe was "removing all subscribe callbacks" as the comment suggests.  This is not true.

rctoris, 2015-03-09:
https://github.com/RobotWebTools/roslibjs/commit/49db59458cbb46e1d71abe0aebb18544301222bb

T045T, 2015-03-15:
I think the root problem is that ```subscribe``` and ```unsubscribe``` do two separate things:
When the user passes a callback, they register or deregister said callback.
But without a callback argument (or when the callback is the first to be added / last to be removed) they actually send the ```subscribe``` or ```unsubscribe``` messages to the rosbridge.

@rctoris , how do you feel about separating this functionality? Or do you think it's enough to point out in the comments?

megawac, 2015-03-15:
The functionality is already separate - just document topic.removeAllListeners

T045T, 2015-03-15:
Yes, the functionality of removing all callbacks is already available, but the current versions of ```subscribe()``` and ```unsubscribe()``` are confusing insofar as they do more than just subscribe to a topic: They register and unregister callbacks as well.
I believe it would be clearer to do away with the parameterless versions of ```subscribe()``` and ```unsubscribe()``` and provide new methods that make clear they *just* unsubscribe the ROS topic without touching the callbacks

rctoris, 2015-03-16:
I vote is for keeping it and making it clear in the comments what is going on. Having a separate unsubscribeX function, whatever it may be called, seems redundant.

zfreedman, 2016-04-22:
This thread is a legend. Just helped out tremendously with me removing compounding listeners. Thanks a ton.

pbeeson, 2020-11-11:
Closing due to age.

closed 2020-11-11 by pbeeson