RobotWebTools/roslibjs #166: "npm install roslib" doesn't bring down runnable code

b3devs, 2015-03-04:
I just installed roslib 0.12.0 (npm install roslib). I then added this to my javascript code: "roslib = require("roslib");", and it fails when I run my web app. Ros.js can't find "./../util/shim/EventEmitter2.js" ... even though it does exist in that location.

There is no node_modules/roslib/roslib.js file and the roblib docs mention building the code, so I looked at this build docs:  https://github.com/RobotWebTools/roslibjs/blob/master/CONTRIBUTING.md

I ran "npm install" from the roslib dir. No problems. I ran "grunt build". It failed:

$ grunt build
A valid Gruntfile could not be found. Please see the getting started guide for
more information on how to configure grunt: http://gruntjs.com/getting-started
Fatal error: Unable to find Gruntfile.

**There is no Gruntfile.js in the npm package.** It appears that it was explicitly excluded from the npm package in .npmignore.

I'm confused how this is supposed to work. I would assume installing roslib via npm would just give me the built roslib.js along with the other npm module depenedencies; but instead I get all of the code and no way to build it. How is this working for other people?

megawac, 2015-03-04:
Alright the issue is `EventEmitter2` isn't included by default in the code generated by browserify. Changing this would be easy but it would result in existing code examples of how to setup `roslib` being incorrect (loading an extra version of `EventEmitter` unnecessarily).

The way to setup `roslib` for browserify is kind of ugly you have to load `EventEmitter2` first and expose it globally as below (this could also be loaded by CDN)

```js
// For roslib
global.EventEmitter2 = require("eventemitter2").EventEmitter2;
```

b3devs, 2015-03-04:
Yep. That did the trick ... but ... ick.

Thanks. I'll close this issue.

closed 2015-03-04 by b3devs

megawac, 2015-03-04:
/cc @rctoris you may want to include `EventEmitter2` to avoid these issues in the future and just change the examples