RobotWebTools/roslibjs #172: JSON parsing in decompressPng

ST-Lab, 2015-03-24:
Dear roslibjs developers.

I found a problem to use roslib.js with compression(png) option.
The function "decompressPng()" in roslib.js, "context.drawImage()" makes
blurry image even if "context.drawImage(image,0,0)".
so that, decompressed data was broken.

- my environment is chrome 41 + android 4.4.2, tegra note 7.
- same issue is https://code.google.com/p/chromium/issues/detail?id=231916

And I also found solution as follows.

- in roslib.js::decompressPng()
- context.imageSmoothingEnabled = false;  (<--- use just before drawImage function)
- context.drawImage(image,0,0);

the method avoids automatic anti-aliasing in chrome+android,
even if the image(decompressed data) has more than 64k pixels.

I hope it could be your help.

rctoris, 2015-03-24:
Thanks for this! Created a patch in https://github.com/RobotWebTools/roslibjs/pull/174

ST-Lab, 2015-03-24:
Thank you, too.

closed 2015-03-24 by ST-Lab