RobotWebTools/roslibjs #173: Service call namespacing is broken

Rayman, 2015-03-24:
Service calls are not namespaced so they will fail. In my Dropbox I've setup a small test case to reproduce this: https://dl.dropboxusercontent.com/u/2603224/namespace_bug.zip

## Setup
```
npm install
roslaunch start.launch
```
## Test
You can run `rosservice list` to verify that both `add_two_ints` and `rosbridge_websocket` run in the same namespace.
```
./call_add_two_ints.js
...
Result for namespaced service call on /test/add_two_ints: 3
```

The normal service call results in rosbridge throwing the following error:
```
[id: call_service:add_two_ints:1] call_service InvalidServiceException: Service add_two_ints does not exist
```

rctoris, 2015-03-24:
I'm confused on the bug. When you namespace something, the service call without the proper namespace should fail, should it not? By placing the node in a namespace, the service call is therefore `/test/add_two_ints` which seems to be working in your example. `/add_two_ints` should not work. 

Perhaps you are thinking this might be a problem with rosbridge not respecting the namespace instead of roslib?

Rayman, 2015-03-24:
Maybe I should clarity a bit. When you call a service *without* a slash at the beginning of the name, that means it should be relative to the current namespace. So for example `add_two_ints` should resolve to `/test/add_two_ints` because the caller is run in the namespace `test`.

A slash at the beginning is an absolute name. So `/add_two_ints` should resolve to just `/add_two_ints`

In the case of my test, both calls should succeed, but one fails so something is wrong. It could of course also be my understanding of the whole namespacing thing.

BTW, topic namespacing does work, I have only trouble with services for some reason.

rctoris, 2015-03-24:
Right, but you cant technically run a roslib client in a namespace since it's not actually running within the ROS echo system. You have run `rosbridge` in the namespace meaning it should be responsible for calling `add_two_ints` appropriately, right?

T045T, 2015-03-24:
Looks like the issue is that rosapi (which provides the ```get_services``` functionality) does not behave differently if run in a namespace. See https://github.com/RobotWebTools/rosbridge_suite/issues/78 for a discussion of this.

In this case, that means ```add_two_ints``` does not resolve, because from the global namespace, it doesn't exist.

As for running a roslib client in a namespace, how about giving ```ROSLIB.Ros``` a ```namespace``` parameter?

rctoris, 2015-03-24:
+1 on a namespace param

Rayman, 2015-03-24:
If possible I don't want my JavaScript code cluttered with namespace code. That should be handled in the launch files.

A solution could be that the rosbridge calls `rospy.get_namespace()` and passes that as a parameter to `rosapi`.

T045T, 2015-04-26:
Closing this, as it was addressed by https://github.com/RobotWebTools/rosbridge_suite/pull/178

closed 2015-04-26 by T045T