RobotWebTools/roslibjs #177: Remove sudo from npm install

rctoris, 2015-04-07:
Currently `npm install` will run `misc/install.sh` which runs some sudo commands. This is unexpected behavior to many and breaks many automated scripts. I propose removing this from the npm install script and placing the instruction to run the script in the README instead.

megawac, 2015-04-07:
Related #133. Ideally we could detect if `libcario` is installed. If it isn't

a) install it (requiring sudo)
b) Error the build and give users instruction on how to install it

jihoonl, 2015-04-09:
I would suggest b) to resolve the issue.  This is issue of `node-canvas` not internal problem of `roslibjs`.

rctoris, 2015-04-09:
+1 to `b`

rctoris, 2015-04-10:
@jihoonl can you run a npm release to propagate this please?

jihoonl, 2015-04-11:
Done.

https://www.npmjs.com/package/roslib

closed 2015-04-11 by jihoonl