RobotWebTools/roslibjs #183: Connecting non ROS tcp client to ROS  robot

MauroPagliarella, 2015-06-30:
Hi all
I'm developing a ground control station to control my robot, (which must run on a computer not running ROS). For this purpose I'm using the Qt tool and I would like to build a tcp client (using the QTcpSocket class), which should connect to a tcp server running on my robot on board computer (which run ROS).  I created a ros_qt package using command catkin_create_qt_pkg, but the tcp server I created using QTcpServer class,  doesn't work. 
Does somebody know if it is possible to use the rosbridge_suite to have a tcp server running on the robot on board computer, so that I can connect my tcp client? If so, is there any C++ client example on how to do this?

Thank you very much.

Rayman, 2015-07-05:
If you want to connect to a rosbridge_server that is running on your robot via TCP you must talk the rosbridge protocol over it. Normally you'd use a library for that such as roslibjs if you use javascript/nodejs.

Why don't you just connect to your robot via the [ROS_MASTER_URI](http://wiki.ros.org/ROS/EnvironmentVariables#ROS_MASTER_URI)? That is way easier, you only need to link to the ros libraries and you can use the normal ROS communication.

MauroPagliarella, 2015-07-05:
Thank you very much for your answer. 
I would like to create my ground control station not using ROS. If I follow you suggestion, I think I have to install ROS on the computer running the GCS, or not?

rctoris, 2015-07-07:
Yes, this would require a ROS library to work. There is a TCP version of rosbridge but I am unaware of an open-source C/C++ version of a rosbridge client library (thought it wouldn't be too hard to implement). There is a Java one for reference: https://github.com/WPI-RAIL/jrosbridge

MauroPagliarella, 2015-08-02:
Thank you for your help. 
Now  I'm  trying to send JSON message to rosbridge but it doesn't work. 
I described in details my attemp in another ticket.  I need to send a std_msgs/int32MultiArray data and I formatted the message as follow:
{\"op\" : \"publish\", \"topic\" : \"/TcMsg\", \"layout\": {\"dim\": [], \"data_offset\": 0}, \"data\":[1, 2, 3, 4, 5]}

Probably I'm sending a wrong formatted string to rosbridge. I tried to change it in many ways but all my attempts failed. 

Do you know which is the correct formatted message to send to rosbridge to publish an array of int32 data?

Thank you very much

MauroPagliarella, 2015-08-02:
In the end my new attemp succeeded. The correct string to rosbridge is:
{\"op\" : \"publish\", \"topic\" : \"/TcMsg\", \"msg\" : {\"layout\": {\"dim\": [], \"data_offset\": 0}, \"data\":[1, 2, 3, 4, 5]}}
Sorry but sometimes is hard to found the mistake.

jihoonl, 2016-02-11:
:smiley:

closed 2016-02-11 by jihoonl