RobotWebTools/roslibjs #184: unable to add two map widget to a page

zenified, 2015-07-03:
When using two map widget on a single page, both widget will not show the map content

rctoris, 2015-07-07:
`roslibjs` is purely a communication library and doesn't have a map widget. Did you mean to post this in `ros2djs` or `ros3djs`?

zenified, 2015-07-07:
Hi Russell
Apology, Yes I mean the ros2djs. I have managed to figured out the issue
that was causing the problem. Following the tutorial "Creating a Basic
Nav2D Widget",

  29     // Setup the nav client.
  30     var nav = NAV2D.OccupancyGridClientNav(...)

The 'new' keyword is missing. Adding the 'new, keyword fixed my issue

29     // Setup the nav client.
  30     var nav = new NAV2D.OccupancyGridClientNav(...)

Sorry for the trouble, shall close the issue
Yu Bin

closed 2015-07-07 by zenified