RobotWebTools/roslibjs #187: Error of roslib install with nodejs version v0.12.7

dwlee, 2015-07-20:
I updated the latest node (v0.12.7) and npm (2.11.3). In this environments, roslib was not complied with following error. 

```
$ npm install roslib

> roslib@0.15.0 preinstall /home/yujin/demos/demo_201502/src/concert_workflow_engine/concert_workflow_engine_blockly/node_modules/roslib
> sh misc/check.sh

libcairo2 dev files found.
 
> canvas@1.1.3 install /home/yujin/demos/demo_201502/src/concert_workflow_engine/concert_workflow_engine_blockly/node_modules/roslib/node_modules/canvas
> node-gyp rebuild

make: Entering directory `/home/yujin/demos/demo_201502/src/concert_workflow_engine/concert_workflow_engine_blockly/node_modules/roslib/node_modules/canvas/build'
  CXX(target) Release/obj.target/canvas/src/Canvas.o
In file included from ../src/Canvas.h:22:0,
                 from ../src/Canvas.cc:7:
../node_modules/nan/nan.h: In function ‘v8::Handle<v8::Value> NanError(const char*)’:
../node_modules/nan/nan.h:252:9: error: ‘New’ is not a member of ‘v8::String’
     fun(v8::String::New(errmsg))
         ^
../node_modules/nan/nan.h:261:12: note: in expansion of macro ‘_NAN_ERROR’
    return  _NAN_ERROR(v8::Exception::Error, errmsg);
....

../src/Canvas.cc: In static member function ‘static v8::Local<v8::Value> Canvas::Error(cairo_status_t)’:
../src/Canvas.cc:545:27: error: ‘New’ is not a member of ‘v8::String’
   return Exception::Error(String::New(cairo_status_to_string(status)));
                           ^
In file included from ../src/Canvas.h:22:0,
                 from ../src/Canvas.cc:7:
../node_modules/nan/nan.h: In instantiation of ‘void NanDispose(v8::Persistent<T, v8::NonCopyablePersistentTraits<T> >&) [with T = v8::Object]’:
../node_modules/nan/nan.h:592:34:   required from here
../node_modules/nan/nan.h:311:5: error: ‘class v8::Persistent<v8::Object>’ has no member named ‘Clear’
     handle.Clear();
     ^
../src/Canvas.cc:546:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make: *** [Release/obj.target/canvas/src/Canvas.o] Error 1
make: Leaving directory `/home/yujin/demos/demo_201502/src/concert_workflow_engine/concert_workflow_engine_blockly/node_modules/roslib/node_modules/canvas/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.13.0-39-generic
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/yujin/demos/demo_201502/src/concert_workflow_engine/concert_workflow_engine_blockly/node_modules/roslib/node_modules/canvas
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 

```

rctoris, 2015-07-20:
related to https://github.com/RobotWebTools/roslibjs/issues/160

jihoonl, 2015-09-02:
Fixed in 0.17.0

https://github.com/RobotWebTools/roslibjs/pull/193

closed 2015-09-02 by jihoonl