RobotWebTools/roslibjs #198: install seems fail

wangsen1312, 2015-09-17:
When I use npm install the roslib
It shows like below:

npm http GET https://registry.npmjs.org/roslib
npm http 304 https://registry.npmjs.org/roslib

> roslib@0.17.0 preinstall /home/xinxinzuo/node_modules/roslib
> sh misc/check.sh

libcairo2 dev files found.
npm http GET https://registry.npmjs.org/aliasify-patch
npm http GET https://registry.npmjs.org/canvas
npm http GET https://registry.npmjs.org/ws
npm http GET https://registry.npmjs.org/xmldom
npm http GET https://registry.npmjs.org/object-assign
npm http 304 https://registry.npmjs.org/canvas
npm http 304 https://registry.npmjs.org/ws
npm http 304 https://registry.npmjs.org/aliasify-patch
npm http 304 https://registry.npmjs.org/object-assign
npm http 304 https://registry.npmjs.org/xmldom
npm http GET https://registry.npmjs.org/browserify-transform-tools
npm http GET https://registry.npmjs.org/commander
npm http GET https://registry.npmjs.org/nan
npm http GET https://registry.npmjs.org/options
npm http GET https://registry.npmjs.org/tinycolor
npm http GET https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/browserify-transform-tools
npm http 304 https://registry.npmjs.org/options
npm http 304 https://registry.npmjs.org/commander
npm http 304 https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/tinycolor
npm http GET https://registry.npmjs.org/nan/-/nan-1.0.0.tgz
npm http 304 https://registry.npmjs.org/nan
npm http GET https://registry.npmjs.org/through
npm http GET https://registry.npmjs.org/falafel

> canvas@1.2.9 install /home/xinxinzuo/node_modules/roslib/node_modules/canvas
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: spawn ENOENT
gyp ERR! stack     at errnoException (child_process.js:988:11)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:779:34)
gyp ERR! System Linux 4.1.5-040105-generic
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/xinxinzuo/node_modules/roslib/node_modules/canvas
gyp ERR! node -v v0.10.25
gyp ERR! node-gyp -v v0.10.10
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! weird error 1
npm http 304 https://registry.npmjs.org/through
npm http 200 https://registry.npmjs.org/nan/-/nan-1.0.0.tgz
npm http 304 https://registry.npmjs.org/falafel
npm http GET https://registry.npmjs.org/foreach
npm http GET https://registry.npmjs.org/acorn
npm http GET https://registry.npmjs.org/isarray/0.0.1
npm http GET https://registry.npmjs.org/object-keys
npm http 304 https://registry.npmjs.org/foreach
npm http 304 https://registry.npmjs.org/object-keys
npm http 304 https://registry.npmjs.org/acorn
npm http 304 https://registry.npmjs.org/isarray/0.0.1
npm ERR! not ok code 0

someone have met it before? Thank you very much

Rayman, 2015-09-18:
I've never saw that error. I foud a possible solution [here](https://stackoverflow.com/questions/21168141/can-not-install-packages-using-node-package-manager-in-ubuntu), could you try installing nodejs-legacy and try again?
```sh
sudo apt-get install nodejs-legacy
```

wangsen1312, 2015-09-18:
@Rayman  I found the mistake, after install the gyp-node  should set python2.7 as the default settings

T045T, 2015-10-20:
@wangsen1312 Does this mean we can close this issue? Or is there a way we can avoid this bug in our config files?

wangsen1312, 2015-10-20:
@T045T   maybe this is not the problem of our part.  you can close the issue. Thank you very much.

closed 2015-10-20 by T045T