RobotWebTools/roslibjs #199: toBlob() to grab webcam images instead of toDataURL()

alexanderkoumis, 2015-09-18:
This might be more of a sensor_msgs question but I'm wondering if the `toBlob()` function currently in FF and Chrome's [Experimental Canvas Features](https://code.google.com/p/chromium/issues/detail?id=67587#c117) can be used to grab the webcam frame instead of `toDataURL()`. As [explained here](https://code.google.com/p/chromium/issues/detail?id=67587#c39) `toBlob()` grabs the compressed jpeg or png's binary data without later expanding it to a base64 string, avoiding a bit of overhead. I'm curious to see if the 'sensor_msgs/CompressedImage' type used to publish an encoded image is meant only for base64-encoded data, or if it works with the binary files as well (will try this weekend).

Rayman, 2015-09-21:
To what code are you refering? [This example](http://wiki.ros.org/roslibjs/Tutorials/Publishing%20video%20and%20IMU%20data%20with%20roslibjs#CA-3fc6413f79f1645f69db70dc902a3a12ecc54f7c_155)?

I think you are able to avoid the overhead you are talking about when you use BSON encoding for rosbridge, with that you can send raw uint8[] over websocket which you can obtain from the toBlob method. To enable this you have to do two things:
- for rosbridge_websocket, set the `binary_encoder` parameter to `bson`
- include [bson.js](https://raw.github.com/mongodb/js-bson/master/browser_build/bson.js) to your html code (roslib will warn if you forget this)

alexanderkoumis, 2015-10-21:
Hi @Rayman, thanks for your response. I'm sending over the binary object but am having trouble finding where the `binary_encoder` parameter is. I've grep'd around and the only mention of this parameter I can find is in `/python2.7/dist-packages/rosbridge_library/internal/message_conversion.py`. Also tried inserting it as a parameter in `/opt/ros/jade/share/rosbridge_server/launch/rosbridge_websocket.launch` and tried doing this in my launch file (feels very wrong):

```xml
<include file="$(find rosbridge_server)/launch/rosbridge_websocket.launch">
    <param name="binary_encoder" value="bson"/>
</include>
```

How can I set this parameter?

Rayman, 2015-10-21:
I think this is all very experimental, because there is not much documentation about this stuff. I discovered this by reading the source.

I think you are supposed to set that parameter file in your launch files just like you did, see [message_conversion.py](https://github.com/RobotWebTools/rosbridge_suite/blob/db17d7afd13d943e98a77d474dec36cf736ee64f/rosbridge_library/src/rosbridge_library/internal/message_conversion.py#L73). I'm also doing that on our robot: [mobile_ui_server.launch](https://github.com/tue-robotics/robot_launch_files/blob/23b55687a24f4ec4833d7ccf54cdad0771d2d3c8/launch/interaction/mobile_ui_server.launch).

If ROS encounters a `uint8[]` or `char[]` in a message, it will encode it using BSON instead of JSON. `sensor_msgs/CompressedImage` data field `uint8[]` so this should work. Note that base64 only increases the size by 37%.

I'm not sure if BSON has been implemented in two-way from rosbridge. Maybe someone else could answer this?
rosbridge --> roslib (experimental)
roslib --> rosbridge (???)

alexanderkoumis, 2015-10-21:
Thanks for your example, still running into a few problems. 

Here is the relevant bit from my launch file (copied from your example):

```xml
    <!-- Launch the websocket server (communication via roslib) -->
        <node name="rosbridge_websocket" pkg="rosbridge_server" type="rosbridge_websocket" output="screen">
            <param name="authenticate" value="false" />
            <param name="port" value="9090"/>
            <param name="address" value=""/>
            <param name="binary_encoder" value="bson"/>
        </node>
```

With the image converted into a Uint8Array, as follows:
```javascript
canvas.toBlob( function( blob ) {
    var reader = new FileReader();
    reader.addEventListener("loadend", function() {
        var byteArray = new Uint8Array( reader.result );
        var imageMessage = new ROSLIB.Message({
            format : 'jpeg',
            data : byteArray
        });
        topics['image'].publish( imageMessage );
    });
    reader.readAsArrayBuffer( blob );
}, 'image/jpeg');
```
rosbridge outputs the following errors:

```
[WARN] [WallTime: 1445460508.844563] Inbound TCP/IP connection failed: field data must be a list or tuple type
[ERROR] [WallTime: 1445460508.859205] [Client 5] [id: publish:/web/input/compressed:29] publish: field data must be a list or tuple type
```

Worth note however I am able get the following to work, and from what I can tell it looks faster (haven't actually done bench-marking) due to the asynchronous image encoding, but the data is still in base64.
```javascript
canvas.toBlob( function( blob ) {
    var reader = new FileReader();
    reader.addEventListener("loadend", function() {
        var imageMessage = new ROSLIB.Message({
            format : 'jpeg',
            data : reader.result.replace("data:image/jpeg;base64,", "");
        });
        topics['image'].publish( imageMessage );
    });
    reader.readAsDataURL( blob );
}, 'image/jpeg');
```

cyberphantom, 2018-03-16:
Hi, maybe my question is little bit basic as I'm new to ROS environment, but I wanna know how to convert the compressedImage in python to CV image on the ROS side as you subscribe to the compressedImage. I found a code in cpp which reads the compressedImage but I'm looking for the python function.
Here is a working code:

`#include <ros/ros.h>
#include <opencv2/highgui/highgui.hpp>
#include <cv_bridge/cv_bridge.h>
#include <image_transport/image_transport.h>

void imageCallback(const sensor_msgs::CompressedImageConstPtr& msg)
{
  try
  {
    cv::Mat image = cv::imdecode(cv::Mat(msg->data),1);//convert compressed image data to cv::Mat

    cv::imshow("view", image);

    cv::waitKey(30);

  catch (cv_bridge::Exception& e)
  {
    ROS_ERROR("Could not convert to image!");
  }
}

int main(int argc, char **argv)
{
  ros::init(argc, argv, "image_listener");
  ros::NodeHandle nh;

  cv::namedWindow("view");
  cv::startWindowThread();

  ros::Subscriber sub = nh.subscribe("/device/image", 1, imageCallback);
  ros::spin();

  cv::destroyWindow("view");
}`

closed 2025-11-08 by EzraBrooks