RobotWebTools/roslibjs #203: "npm install roslib" fails on Windows

mabuchner, 2015-10-16:
"npm install roslib" fails on Windows (cmd.exe) with the following error message:
```shell
'sh' is not recognized as an internal or external command, operable program or batch file.

npm ERR! roslib@0.17.0 preinstall: `sh misc/check.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the roslib@0.17.0 preinstall script.
```

I only get the following output using Cygwin64:
```shell
$ npm install --save roslib
/cygdrive/c/Program Files/nodejs/npm: line 2: $'\r': command not found                    
/cygdrive/c/Program Files/nodejs/npm: line 3: dirname: command not found                  
/cygdrive/c/Program Files/nodejs/npm: line 4: $'\r': command not found                    
/cygdrive/c/Program Files/nodejs/npm: line 5: syntax error near unexpected token `$'in\r''
'cygdrive/c/Program Files/nodejs/npm: line 5: `case `uname` in                            
```

Rayman, 2015-10-16:
This will be fixed in the next release (0.18). For now you can install the developement version directly from GitHub:
```
npm install --save git+https://github.com/RobotWebTools/roslibjs.git#develop
```

mabuchner, 2015-10-19:
Ok, thanks, this seems to resolve the initial problem.

Now I get the following compilation error:
.\node_modules\roslib\node_modules\ws\node_modules\bufferutil\node_modules\nan\nan.h(459): error C2039: 'SetCounterFunction' : is not a member of 'v8::Isolate' [.\node_modules\roslib\node_modules\ws\node_modules\bufferutil\build\bufferutil.vcxproj]

This is probably caused by the fact, that I'm forced to use an old version of node.js (v0.11.13).

Rayman, 2015-10-19:
I just tested this on v0.11.13. If you look a bit below in the install log, you can see that this is just an optional dependency that fails to install. The install succeeds and roslib should just work.
```
npm WARN optional dep failed, continuing bufferutil@1.2.1
```

mabuchner, 2015-10-19:
Yes, it's working. I'm sorry for the trouble. All those red error messages made me assume, that it won't work.

closed 2016-02-11 by jihoonl