RobotWebTools/roslibjs #215: tfClient doesn't seem to transform scene with ros3Djs

Driptap, 2016-02-26:
When visualizing a pointCloud2 or marker using ROS3Djs, subscribing to a tf server with the tfClient in roslibjs doesn't seem to make any transforms to the resulting rendered image. Everything seems to be shown in the same frame, ignoring what is subscribed to by the tfClient

rctoris, 2016-02-26:
This should work -- everything should be transformed into the frame which tfClient specifies as the base frame. Does the example here https://github.com/RobotWebTools/ros3djs/blob/develop/examples/interactivemarkers.html work? You can change to another of the available frames in that example and everything should transform. https://github.com/RobotWebTools/ros3djs/blob/develop/examples/interactivemarkers.html#L35

Do you have an example to provide?

Driptap, 2016-02-27:
Hey @rctoris thanks for your reply, turns out we didn't have tf web re-publisher running. @T045T pointed it out!

closed 2016-02-27 by Driptap