RobotWebTools/roslibjs #216: Deprecated CanvasRendering Message

DLu, 2016-03-03:
In src/util/shim/decompressPng.js: 
'CanvasRenderingContext2D.webkitImageSmoothingEnabled' is deprecated. Please use 'CanvasRenderingContext2D.imageSmoothingEnabled' instead.

T045T, 2016-03-04:
from https://github.com/RobotWebTools/roslibjs/blob/develop/src/util/shim/decompressPng.js#L35-L37 :
```javascript
    context.imageSmoothingEnabled = false;
    context.webkitImageSmoothingEnabled = false;
    context.mozImageSmoothingEnabled = false;
```

The new method is used along with the browser-specific old ones - I think it's a good idea to leave those in case people are using an older browser for whatever reason. We should probably even add `msImageSmoothingEnabled` for IE.

DLu, 2016-03-04:
That's fair.

closed 2016-03-04 by DLu