RobotWebTools/roslibjs #217: maxEventListeners is not a function

spaghetti-, 2016-03-07:
I'm using the latest release with node (tested with 4.0 and 5.4) and I'm getting the following error:

`Uncaught TypeError: this.setMaxListeners is not a function`

Is there anything I'm missing? Even if I remove this line, `this.connect()` seems to fail indicating that something else is not loading/exporting as expected.

peterpolidoro, 2016-11-07:
I get the same error:

main.bundle.js (line 36574)
ORIGINAL EXCEPTION: this.setMaxListeners is not a function

I am trying to use roslibjs with node v6.9.1 and TypeScript.

T045T, 2016-11-07:
Looks like you might be missing the ```new``` keyword:

```
$ node
> var roslib = require('roslib');
> var foo = roslib.Ros();
TypeError: this.setMaxListeners is not a function
[...]
> var asdf = new roslib.Ros();
undefined
> asdf
RosTCP {
  encoding: 'utf8',
  socket: null,
  idCounter: 0,
  isConnected: false,
  groovyCompatibility: true,
  _events: { maxListeners: 0 },
  _conf: { maxListeners: 0 } } 
```

peterpolidoro, 2016-11-07:
That works for me, thanks!!

vkuehn, 2017-03-02:
so that issue can be closed right ?

T045T, 2017-03-06:
Looks like it :)

closed 2017-03-06 by T045T