RobotWebTools/roslibjs #218: Upgrade EventEmitter2 to v1.0.0?

eadlam, 2016-03-26:
I'm using roslib in a project with webpack. It breaks when it tries to import roslib, I think because roslib uses eventemitter2 v0.4.13 which uses broken amd module styles. There are some known workarounds for webpack, but I think the eventemitter2 shim in roslib is preventing those workarounds from working.

I tried upgrading eventemitter to v1.0.0 and removing the shim and that seemed to fix the problem for me. I also ran the `grunt test` task in roslib and all the tests passed. 

I'm wondering if roslib is actually dependant on v0.4.13, and if not can we upgrade it to v1.0.0 and remove the shim?

eadlam, 2016-04-07:
Submitted pr #221

jjclark1982, 2017-09-28:
It's possible to work around the broken shim to use roslib 0.18.0 with webpack. Set the following alias in `webpack.config.js`:

```js
    resolve: {
        alias: {
            eventemitter2: require('path').resolve(__dirname, 'node_modules/eventemitter2'),
        }
    }
```

Rayman, 2019-09-23:
The latest version has `"eventemitter2": "^4.1.0"` so this issue can be closed I think.