RobotWebTools/roslibjs #219: Callback in subscriber not triggered

botonchou, 2016-03-28:
I tried to run `examples/simple.html` but there's no message received from topic `/listener`.
I expect something like `Received message on /listener: Hello, World` will show up in the chrome console. However, the only things I got is
```
Connection made!                                                        simple.html:24
```
I tried to insert `debugger` inside the callback function, and it's not triggered.

I followed every steps from *1. roscore* to *5. roslaunch rosbridge_server rosbridge_websocket.launch*.

On the server side, everything looks fine.
```
SUMMARY
========

PARAMETERS
 * /rosbridge_websocket/address: 
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rosbridge_websocket-1]: started with pid [5266]
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [WallTime: 1459132904.316445] Rosbridge WebSocket server started on port 9090
process[rosapi-2]: started with pid [5288]
[INFO] [WallTime: 1459132917.817918] Client connected.  1 clients total.
[INFO] [WallTime: 1459132918.533898] [Client 0] Subscribed to /listener
```

Not sure whether I ran this in the correct way or not...
Any idea how to solve this problem? Thanks

rctoris, 2016-03-31:
Are you running `rostopic pub /listener std_msgs/String "Hello, World"` after you start rosbridge and load the example page?

botonchou, 2016-03-31:
You mean `roslaunch rosbridge_server rosbridge_websocket.launch`?

I tried running `rostopic pub /listener std_msgs/String "Hello, World"` before and after `roslaunch rosbridge_server rosbridge_websocket.launch`. Both not working. In fact, I tried all the combinations but none of them works : (

Did this ever happen to other people before? I guess I might have missed something...

wintermuteger, 2016-04-11:
I am facing the identical issue with my own code (so not the example). The rosbridge reports a connection and a subscription, and with rostopic echo I can verify that s.th. is published. I am receiving no callback though in my JS...

rctoris, 2016-04-11:
Are you running the current development branch? Have you tried with with the released version? This will help narrow down when a bug may have been introduced: http://cdn.robotwebtools.org/roslibjs/current/roslib.js

botonchou, 2016-04-11:
It was on develop branch. But I just tried the master branch and the link you gave to me `http://cdn.robotwebtools.org/roslibjs/current/roslib.js`, both are not working. same as before

Are you able to reproduce this issue?

wintermuteger, 2016-04-12:
I am running rosbridge_suite-0.7.13 and I am linking to the current/roslibjs. As eventemitter I am linking to 0.4.14.

wintermuteger, 2016-07-17:
Today I finally managed to come back to this topic... I did some debugging, but am running out of ideas.

Connecting and subscribing does actually work (as I can see in the server log and in during JavaScript debugging). Unfortunately it seems that even the lowest level (is it really lowest?) is failing. I tried to break on the "onmessage" callback in the SocketAdapter, but even this one does not trigger.

So summed up, it seems that the WebSocket makes connection to the rosbridge and the rosbridge is registering. It is also possible to subscribe to a topic (type std_msgs/Float32). So PC to Robot direction works.

After that I would expect that the topic is forwarded to rosbridge. But it is either not leaving the robot (with the right address) or not received correct ...

I tried some basic port stuff (checking WebSocket connections with a webtool over different ports, changing rosbridge to the port which is working correctly). Is there anything which might be configured for the WebSocket?

Could it be related to the fact I am using Indigo rosbridge?

wintermuteger, 2016-07-24:
I started debugging on rosbridge and it turns out that there is an open issue: "Rosbridge only one way on OSX  #198". Sounds very close to what I/we observed. Direction to the bot works fine, but no response is triggered as the installed bson (I didn't use rosdep or any other ros tool but installed manually) is not compatible to the required one.

With this finding my proposal would be to close this issue (not roslibjs related) but maybe add a note in the mainpage that points out to people that bson module might be a pitfall, which leads to an error that is difficult to debug (as one direction works without problems and only the reverse direction fails silently).

Unfortunately I am not developing on rosbridge myself, but maybe it could check if the required option is in BSON available and if not trigger an error/warning. Could help some peoples time on debugging.

ilidar, 2016-08-11:
Any ideas what happening ?

rasheinstein, 2016-09-23:
I found the problem. If you installed rosbridge from source which you probably did for kinetic because it came out late, you may have installed a package named 'bson' using `pip install bson`. You should do a `pip uninstall bson` and then uninstall and reinstall the rosbridge_suite. Rosbridge-suite is now available on kinetic

clynamen, 2016-11-01:
I have the same problem on archlinux. 
I tried many versions of bson with pip but the problems remain. 

What is the version required by roslibjs/rosbridge_server?

T045T, 2016-11-07:
On debian systems there's an ```apt-get``` package called [python-bson](http://packages.ubuntu.com/xenial/python-bson), which contains MongoDB's BSON implementation. That is the one rosbridge_server uses. Looks like the equivalent Arch package to get that implementation is [PyMongo](https://www.archlinux.org/packages/community/i686/python-pymongo/files/).

T045T, 2018-04-16:
Several people have posted a solution to this problem:

They had installed the wrong `bson` module. If you install dependencies using `rosdep` or use the debian package, you should not run into this particular issue.

If this doesn't fix your problem, please open a new issue describing it in more detail.