RobotWebTools/roslibjs #222: any requests don't respond.

ebugfix, 2016-04-11:
Hello, I'm using roslibjs by bower and rosbridge-suite compiled from source code (develop branch).

I'm able to connect to the server and subscribes any topics all right, this is shown by ros console, 

but when I try to do any requestes to bridge-websocket server this don't works. 

e.g. 
```js
console.log(ros.isConnected) // returns true
 ros.getNodes(function (nodes) {
        angular.forEach(nodes, function (name) {
          data.nodes.push({ name: name });
   });
});
console.log(ros.isConnected) // returns true
```

getNodes never returns and data are always empty, this happens with all functions getNodes, getTopics, getServices, getParams and so on.

has anyone that can help me ?

thanks

cfpperche, 2016-04-11:
when I call rosapi by console this works fine,
vilma@vilma:~/vilma_ws$ rosservice call /rosapi/service_type "service: '/rosapi/services'" 

returns
type: rosapi/Services

but when, e.g., ros.getNodes, it yet returning undefined and callbacks are not called.

the connection between roslibjs and websocket is established.

cfpperche, 2016-04-12:
I try to do the [BasicRosFunctionality](http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality) tutorial from ros webpage, but the mistakes still happens.

[the gist of my example](https://gist.github.com/cfpperche/a9aeb73462e25e9327608f5a80bd4157)

rosbridge-suit branch develop

cfpperche, 2016-04-12:
sorry, the problem was happens because I install python-bson with 

```sh
pip install bson --upgrade
```
and this cause conflicts that, do 
```sh
pip remove bson
```
all works fine.

closed 2016-04-12 by ebugfix