RobotWebTools/roslibjs #225: Cannot get defined constants from ros messages with getMessageDetails()

devguy22, 2016-04-27:
We have several ros messages that all have defined constants like test.msg:
string TYPE_ORANGE=orange
string TYPE_RED=red
std_msgs/Header header
   uint32 seq
   time stamp...

I can get at all of the non-constant information, but I need those too.  Accessing via python uses a dot notation like test.TYPE_ORANGE or c++ like test::TYPE_ORANGE.  Can getMessageDetails be modified to access those?

BogdanAlexandru, 2017-03-01:
I can look into this if contributions are welcome. Pinging @jihoonl for opinion.

jihoonl, 2017-03-02:
Um attaching constants in all messages will create lots of overheads. rospy and roscpp are not doing it either. They just read class structure directly from file.

But I agree that it is missing feature. How about extending `/rosapi/(message, service_request, service_response)_details` srvs to provide constants?

Alqio, 2022-11-16:
Any updates on this?

MatthijsBurgh, 2022-11-16:
Fixed by https://github.com/RobotWebTools/rosbridge_suite/pull/412

User needs to provide his/her own callback function which reads the service response with the msg/srv details.

closed 2022-11-16 by MatthijsBurgh