RobotWebTools/roslibjs #236: get CannotSendRequest,ResponseNotReady problem!

twxjyg, 2016-09-03:
When I want to get all Topics and Params use getParams,getTopics method, I got this ERROR log in rosbridge_server log:

```
[ERROR] [WallTime: 1472880684.868432] Error processing request: 
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/private/Documents/bazel_ws/catkin_ws/src/rosbridge_suite/rosapi/scripts/rosapi", line 67, in get_topics\n    return TopicsResponse(proxy.get_topics())\n', '  File "/home/private/Documents/bazel_ws/catkin_ws/src/rosbridge_suite/rosapi/src/rosapi/proxy.py", line 50, in get_topics\n    return [x[0] for x in get_published_topics()]\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 376, in get_published_topics\n    code, msg, val = get_master().getPublishedTopics(namespace)\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 106, in wrappedF\n    return f(*args, **kwds)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__\n    return self.__send(self.__name, args)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request\n    verbose=self.__verbose\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request\n    return self.single_request(host, handler, request_body, verbose)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1298, in single_request\n    self.send_request(h, handler, request_body)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1400, in send_request\n    connection.putrequest("POST", handler, skip_accept_encoding=True)\n', '  File "/usr/lib/python2.7/httplib.py", line 877, in putrequest\n    raise CannotSendRequest()\n', 'CannotSendRequest\n']
[ERROR] [WallTime: 1472880684.869038] Error processing request: 
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/private/Documents/bazel_ws/catkin_ws/src/rosbridge_suite/rosapi/scripts/rosapi", line 157, in get_param_names\n    return GetParamNamesResponse(rosapi.params.get_param_names())\n', '  File "/home/private/Documents/bazel_ws/catkin_ws/src/rosbridge_suite/rosapi/src/rosapi/params.py", line 84, in get_param_names\n    return rospy.get_param_names()\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 480, in get_param_names\n    code, msg, val = _param_server.getParamNames() #MasterProxy does all the magic for us\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 106, in wrappedF\n    return f(*args, **kwds)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__\n    return self.__send(self.__name, args)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request\n    verbose=self.__verbose\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request\n    return self.single_request(host, handler, request_body, verbose)\n', '  File "/usr/lib/python2.7/xmlrpclib.py", line 1303, in single_request\n    response = h.getresponse(buffering=True)\n', '  File "/usr/lib/python2.7/httplib.py", line 1039, in getresponse\n    raise ResponseNotReady()\n', 'ResponseNotReady\n']
[ERROR] [WallTime: 1472880684.869482] [Client 23] [id: call_service:/rosapi/topics:4] call_service ServiceException: service [/rosapi/topics] responded with an error: error processing request: 
[ERROR] [WallTime: 1472880684.870944] [Client 23] [id: call_service:/rosapi/get_param_names:3] call_service ServiceException: service [/rosapi/get_param_names] responded with an error: error processing request:
```

But I found something intresting:
This code make problem 90%
```
//ROSInstance is Ros instance,  ROSParams,ROSTopics is my global var
  ROSInstance.getParams(function(params) {
    console.log('ROS Param List:')
    console.log(params)
    ROSParams = params
  })
  ROSInstance.getTopics((topics)=>{
    console.log('ROS Topic List:')
    console.log(topics)
    ROSTopics = topics
  })
```
This code work nice:
```
new Promise((resolve, reject)=>{
  ROSInstance.getParams(function(params) {
    console.log('ROS Param List:')
    console.log(params)
    ROSParams = params
    resolve()
  })
}).then(()=>{
   ROSInstance.getTopics((topics)=>{
    console.log('ROS Topic List:')
    console.log(topics)
    ROSTopics = topics
  })
})
```
The problem may be about lock, i guess.

twxjyg, 2016-09-11:
Any body come to confirm this problem?

jihoonl, 2016-09-11:
I don't understand the problem. Could you explain the problem in detail?

twxjyg, 2016-09-12:
Ok, I want get all published topic and registed param on master in my client application, and I call getTopics and getParams this two method after I made a Ros instance, but I can not get value callback most time, after I use a "Promise" to get Params firstly, and get Topics secondly, I can get the data every time. And I take a look at my rosbridge_server log, I can see some error log when I did not use the "Promise".