RobotWebTools/roslibjs #243: No error handling mechanism provided.

chpatton013, 2016-12-15:
# Problem description

I'm building a node application that connects to a rosbridge websocket server. I'm supplying data to the application by playing back a log file on the same machine.

I ran into the following problem with one bag file:

```
SyntaxError: Unexpected end of JSON input
    at WebSocket.onMessage (.../app/node_modules/roslib/src/core/SocketAdapter.js:104:28)
    at WebSocket.onMessage (.../app/node_modules/ws/lib/WebSocket.js:418:14)
    at emitTwo (events.js:106:13)
    at WebSocket.emit (events.js:191:7)
    at Receiver.ontext (.../app/node_modules/ws/lib/WebSocket.js:816:10)
    at .../app/node_modules/ws/lib/Receiver.js:477:18
    at Receiver.applyExtensions (.../app/node_modules/ws/lib/Receiver.js:364:5)
    at .../app/node_modules/ws/lib/Receiver.js:466:14
    at Receiver.flush (.../app/node_modules/ws/lib/Receiver.js:340:3)
    at Receiver.finish (.../app/node_modules/ws/lib/Receiver.js:482:12)
```

The error occurs within roslibjs at [this line](https://github.com/RobotWebTools/roslibjs/blob/d2ce7d1cc14ebd2a4a03dd2a588c239b4fc843dd/src/core/SocketAdapter.js#L104).

The JSON parse is failing because rosbridge sent malformed data. I believe this happened because the bag file has one or more corrupted messages. However, this could also happen in real-world situations if we were to approach our network-bandwidth- or compute-budgets.

Using a top-level exception handler to capture this failure is not a great solution because:
* we have to infer the context for the failure based on the error message in the exception
* we don't know what state any part of the system is left in
* we end up catching errors that shouldn't be caught
* the only safe thing to do at this point is terminate the process

A single corrupted message is not a good enough reason to crash an entire application.

# Feature request

roslibjs should provide a means to attach error callbacks for failures to parse incoming messages. You should be able to attach three different types of error callback:
* a subscriber-level basis (provided as another parameter to `subscribe`)
* a global callback that is invoked for any error that does not have a subscriber-level callback
* a global callback that is invoked for all errors, even if they have a subscriber-level callback

jihoonl, 2017-01-17:
I will be happy to accept a pull request for new feature :)

EzraBrooks, 2025-11-08:
> The JSON parse is failing because rosbridge sent malformed data. I believe this happened because the bag file has one or more corrupted messages. However, this could also happen in real-world situations if we were to approach our network-bandwidth- or compute-budgets.

I do not think this is due to bag file corruption, and I do not think this can be caused by network bandwidth issues. The messages are not multipart content (with the exception of when fragmentation is in use, in which case they are validated that all parts are received).

There's an open issue on rosbridge_suite to find the bug in JSON serialization causing this issue. Given that this is a violation of the API contract, I don't think we need to build additional fallbacks and failing hard is the correct solution.