RobotWebTools/roslibjs #244: uint8[] received as string

yesennes, 2016-12-21:
Using roslibjs, I subscribed to an sensor_msgs/Image message and attempted to use the data, which should have been uint8[]. I was somewhat surprised to find it was a string. I understand that a string is a char[] and a char is a uint8, so I tried to use charCodeAt to interpret the string as a uint8 array. However, it doesn't seem to be a direct correspondence.

It seems to be consistent, that is the same uint8[] always comes out as the same string, but I couldn't figure out how to get the original data back.

jihoonl, 2017-01-17:
uint8[] from rosbridge server is encoded as b64 base. so you may need to decode to get a original message.

Solrac3589, 2018-01-11:
If someone search a fast solution, I managed to decode it in 2 steps.

lets imagine, our uint8[] is result.pose:

1- var p = atob(result.pose);

With this, the variable is decoded in strange chars. In my case i was watching triangles.

2- p.charCodeAt(3)

This way the char is decoded to a int again. I did several tries and it works.

If you want to have the full array full decoded, I'm just thinking about do a for loop.