RobotWebTools/roslibjs #254: Creating a New Message Type.

sradmard, 2017-05-10:
Hi,

I was wondering how I can create a new message type that can be recognised by roslibjs. I need to create a new message type and be able to access to that through roslib. 

Thank you so much in advance for any help.

marcoarruda, 2017-05-10:
I think I had a problem like this before.
After creating a new message and compiled it inside a robot, my rosbridge server wasn't able to recognize this new message, when trying subscribe a topic.

If you are working with more than 1 computer (like I was), you must compile/generate this new message in the publisher side (e.g: your robot) and also in the rosbridge server computer (e.g: ground station), which is subscribing to it.

jihoonl, 2017-05-11:
@marcoarruda's answer is correct. The PC which runs rosbridge_server should have the newly created message type. `roslibjs` is just a client that receives JSON data that is formatted by rosbridge server.

closed 2017-05-11 by jihoonl