RobotWebTools/roslibjs #257: Allow asynchronous service callbacks

sjlevine, 2017-06-21:
Hello,

First, let me thank you for your hard work on this wonderful project!

When advertising a service with roslibjs, the service callback returns a boolean value and fills out a response. Immediately when the callback returns, this response is sent back to the service callee. This model, while fine for quick service callbacks like add_two_ints, however prevents service callbacks that take time and must run asynchronously. The response will be send back immediately, before the asynchronous code finishes.

As a result, a service callback could not (for example) call other services as part of its own service callback. If the callback currently tried to call other services, they do not finish in time before the response must be sent.

I'm working on a patch for this using the Javascript Promise API; I will have more details shortly.

Thanks!

Saraujoc92, 2019-05-21:
I know this is old, but according to the [Official Documentation](http://wiki.ros.org/ROS/Patterns/Communication),  
> (Services) Should be used for remote procedure calls that terminate quickly
If you want to invoke long running processes, I would advise looking into the Action pattern. Under the hood, it's just a topic based protocol to send goals, feedback, and a result.

closed 2021-06-13 by MatthijsBurgh

smihael, 2022-03-29:
Although actions are preferred ROS pattern for calls that invoke long running processes, in certain occasions services are the easiest alternative, if you want to implement certain functionality using a language that is not supported by actionlib.  

A possible workaround to send multiple service calls in a row is by following [the premises pattern](https://sebhastian.com/javascript-wait-for-function-to-finish/).

Example that worked for me:

```js

var slow_client = new ROSLIB.Service({
    ros: ros,
    name: '/slow_service',
    serviceType: 'std_srvs/Trigger'
});

async function call_slow(i) {
    return new Promise(resolve => {
        var slow_request = new ROSLIB.ServiceRequest({string: 'test'+i});
        slow_client.callService(slow_request, result => { 
          console.log(result.message);
          resolve();
        });
    });
}

// send ten requests to slow_client
async function slow() {
    for (var i = 0; i < 10; i++) {
        await call_slow(i);
    }
}
```