RobotWebTools/roslibjs #268: How to import gennodejs generated message to roslibjs?

susu3621, 2017-10-21:
We have several customized message, and planning to using rosbridge to Windows user.
However, when I try to import gennodejs to roslibjs, I find it's not so easy.

As ros build result always has gennodejs folder from message_generator, I thought roslibjs is using the same mechanism of message and service.

My question is: 
Is roslibjs designed to compatible with gennodejs result?
Is there a typical way or tutorial for adding customer message into roslibjs? 

Thanks a lot!

jihoonl, 2017-10-30:
I didn't check the roscon video yet. Can @nuclearsandwich or @sevenbitbyte answer this?

nuclearsandwich, 2017-10-31:
Sadly Chris's talk was one of the only ones I missed at ROSCon. Here is the [video](https://vimeo.com/236189788) in question. I think it would be a good idea to cross-post this question to the [gennodejs issue tracker](https://github.com/RethinkRobotics-opensource/gennodejs/issues) and get more feedback from them. @sevenbitbyte will almost certainly have a better handle on it than I will but I was under the impression that different message types are handled by roslib as plain JSON objects and the typed message support provided by gennodejs isn't needed at this layer. But I am very new to the project and so may not understand.

susu3621, 2017-11-25:
Thanks Guys! I dig some more and I agree with @nuclearsandwich now. Actually we don't need gennodejs at this layer. Seems plain JSON objects are more flexible and enough here. I think we can close this issue now.

closed 2017-11-25 by susu3621