RobotWebTools/roslibjs #272: Topic Types does not match : [Sensor_msgs/Image] vs [sensor_msgs/CompressedImage]

pushkalkatara, 2017-11-06:
Referred this tutorial  -
http://wiki.ros.org/roslibjs/Tutorials/Publishing%20video%20and%20IMU%20data%20with%20roslibjs
topics type not match error. 
http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/CompressedImage.html
http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/Image.html

Is this tutorial synced with updates?

pushkalkatara, 2017-11-06:
Does CompressedImage take only JPEG as input?

jihoonl, 2017-11-07:
This tutorial seems to have been created by someone in the community. But, I think it should work anyway. What do you mean by topics type not match error?

> Does CompressedImage take only JPEG as input?

[sensor_msgs/CompressedImage](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/CompressedImage.html) can accept either jpeg or png. See the format description

pushkalkatara, 2017-11-09:
[image]
[image]

pushkalkatara, 2017-11-19:
Solved this issue recently, imageview was not able to display the image but a simple opencv workaround worked.
```python        np_arr = np.fromstring(ros_data.data, np.uint8)
        image_np = cv2.imdecode(np_arr, cv2.IMREAD_COLOR)
        cv2.imshow('cv_img', image_np)
        cv2.waitKey(2)
```

Though the video is lagging and saturating the rosbridge. Please suggest alternatives.

cyberphantom, 2018-03-15:
Any solution to this problem yet?
 @pushkalkatara I tried your suggested imdecode in the callback function to subscribe to the compressed Image but I'm getting an  error

`    self.frame = cv2.imdecode(np.frombuffer(data, cv2.IMREAD_COLOR), 1)
TypeError: data type not understood
`

T045T, 2018-03-26:
Are you sure the bridge is saturated?
The example you linked only sends camera frames (at 640 pixels width) every 250ms. That shouldn't be enough to cause lag, but it *is* just 4fps, which might be why it seems laggy.

closed 2021-06-13 by MatthijsBurgh