RobotWebTools/roslibjs #274: robotwebtools cdn is broken (gone?)

minshallj, 2017-11-16:
It looks like http://cdn.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js and http://cdn.robotwebtools.org/roslibjs/current/roslib.min.js (and all of cdn.robotwebtools.org) no longer work. It would be nice to get an official CDN set up again. This would hopefully include changing the roslibjs tutorial at http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality

keego, 2017-11-18:
Just ran into this issue today when attempting the [Simple Interactive Markers Demo](http://robotwebtools.org/demos/im.html).

**Details**

 Console from http://robotwebtools.org/demos/im.html:

[image]

Response from http://cdn.robotwebtools.org/threejs/current/three.min.js:

[image]

hoangthien94, 2017-11-19:
I met the same issue when trying the [roslibjs tutorial](http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality).
I can solve it temporarily by downloading the requested files, in my case [roslib.js](https://github.com/RobotWebTools/roslibjs/tree/develop/build) and [eventemitter2.min.js](https://github.com/asyncly/EventEmitter2/tree/master/lib), place it locally, change the directory in the html file accordingly. After that I can run the example.
I don't know if the replacement files I use have the same content with the original files from cdn.robotwebtools.org or not (I haven't seen them before), so I can't say about their compatibility with other examples.
I think it's best if the necessary librarries are included in the repo.

jihoonl, 2017-11-20:
CDN and all server are migrating to the new server at the moment. This issue will be resolved soon. I suggest to follow google mailing list

https://groups.google.com/d/msg/robot-web-tools/PlW1ECoL91Y/uI80cp3zDAAJ

jihoonl, 2017-11-20:
I will keep it open until it gets fixed.

jihoonl, 2017-12-07:
It is fixed now. Please refer this e-mail.

https://groups.google.com/d/msg/robot-web-tools/plGsZWwtNv8/1M3QCCV3AgAJ

closed 2017-12-07 by jihoonl

minshallj, 2017-12-07:
Thanks for that link :) Now to update everywhere cdn.robotwebtools.... is referenced! I issued a pull request (above) and requested edit permission for http://wiki.ros.org/action/login/roslibjs/Tutorials/BasicRosFunctionality but if anyone here already has edit permission feel free to `s/cdn.robotwebtools/static.robotwebtools/g` that page

minshallj, 2023-04-27:
It looks like the entire robotwebtools.org website is down now :( not sure if there's a migration of where people are supposed to go now

MatthijsBurgh, 2023-04-27:
How about https://robotwebtools.github.io/