RobotWebTools/roslibjs #283: Issues with very large number arrays

uahic, 2018-02-05:
I encountered an issue when sending out a nav_msgs/OccupancyGrid message from roslib to ROS. If the data array is fairly small (several thousands entries) then it works fine, but in the order of 4 million entries it throws Errors that the data array has 'noneType'. It probably has to do with encoding but if the issue is something known and different please let me know.

Thanks!

Behery, 2018-02-06:
Maybe it has to do with [json decode limit](https://docs.python.org/2/library/json.html#implementation-limitations)? Can you provide a minimal example so I can test this?

uahic, 2018-02-08:
I am using Typescript (along with Angular 5), do you have typescript ready?

uahic, 2018-02-08:
Hm same issue with services, now I really do need some help :/

closed 2018-02-08 by uahic

jihoonl, 2018-02-09:
Which browser are you using?

uahic, 2018-02-09:
It was a bug in my code actually but first I though I really made sure it is not, obviously I was wrong.
It works now but unfortunately it seems the JSON pipeline is extremely slow for large arrays. For my prototype stuff its okay but I might need to send the data via another communication layer and inject it to ROS than using rosbridge

jihoonl, 2018-02-09:
yes. You might want to try `compression` or `bson` encoding

uahic, 2018-02-09:
do you have an example how to use bson for sending some message roslibjs -> rosBridge?  I did see a chunk of code for the other direction in the source code but now how to activate that