RobotWebTools/roslibjs #289: Add support for dynamic_reconfigure

rangeli, 2018-05-11:
The dynamic_reconfigure package uses service calls to provide a generic interface to change a node's parameters.  Currently, a roslibjs client can interface to a node implementing dynamic_reconfigure using a service call (See [dynamic_reconfigure/Reconfigure.srv](http://docs.ros.org/jade/api/dynamic_reconfigure/html/srv/Reconfigure.html)).

However, it would be good to incorporate a Javascript API within roslibjs to interface with dynamic_reconfigure (similar to their [Python API](http://docs.ros.org/api/dynamic_reconfigure/html/dynamic_reconfigure.client.Client-class.html)).

rangeli, 2018-05-14:
I've created a tutorial wiki on how to currently use roslibjs as a dynamic_reconfigure client through service calls found [here](http://wiki.ros.org/roslibjs/Tutorials/Writing%20a%20dynamic_reconfigure%20client%20with%20roslibjs).

EzraBrooks, 2025-11-08:
no new ROS 1 features will be added to roslibjs, as ROS 1 is EOL.