RobotWebTools/roslibjs #290: The listener callback in subscribe is not called

Spijkervet, 2018-05-21:
As described earlier in https://github.com/RobotWebTools/roslibjs/issues/219, I am still having issues getting console log outputs from the listener callback by following the tutorial. I am currently running on the ros-bridge server provided by the apt-get package ros-kinetic-rosbridge-server. My roslib is on the currently released version on the npm (master branch, I assume).

Any ideas what is wrong? I have done all standard installation procedures.

T045T, 2018-05-21:
Can you give us a bit more information on your problem?

First, what tutorial are you following? You mention npm, but I don't think there is a Node tutorial yet.

Second, "all standard installation procedures" is a bit vague. What have you done exactly?
Things like the setup of your command line (which setup.bash file you sourced) may make a difference.

That said: If you're using one the `simple.html` example from this repo or the tutorial code from http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality , the console should print "Connection made!" before anything else.

If that's not happening, your problem is probably in the setup.
If it is, sharing the extra information I asked for may allow us to help you.

Spijkervet, 2018-05-21:
I tried everything on a clean Ubuntu install on a virtual machine, and that seems to work fine. Still not sure what is going wrong in my own setup.

Spijkervet, 2018-05-21:
Update: You can close this. I found out I had a different tornado version installed on my global python installation. All works fine! :)

T045T, 2018-05-21:
That's a relief!

From your description (thanks for all the details!) I was stumped - I have a Xenial VM where I tried the same thing and couldn't reproduce the error.

Glad you figured out a solution!

closed 2018-05-21 by T045T

benjaminbecker, 2018-05-27:
I have the same issues as described above. How can I detect and fix the problem with the different tornado version?

Spijkervet, 2018-05-28:
Make sure to `pip uninstall tornado` and check (e.g. in your site-packages folder) that it is completely removed.

After that, re-run `apt-get install ros-DISTRO_VERSION-rosbridge-server` to get the correct version of Tornado installed.

benjaminbecker, 2018-05-28:
Thank you! That solved the problem within 10 seconds after 1 hour of debugging my code.

meseta, 2018-07-25:
This fixed our issue too, thanks!