RobotWebTools/roslibjs #291: Question: syntax to retrieve topic type

rachelspykerman, 2018-06-21:
Hi,

I'm trying to retrieve the topic type for a particular topic name. I can get the topics, but unable to find documentation on how to get the topic type. I tried the following, but this does not work:

    function getTopics() {
      var topicsClient = new ROSLIB.Service({
        ros : ros,
        name : '/rosapi/topics',
        serviceType : 'rosapi/Topics'
      });

      var request = new ROSLIB.ServiceRequest();
      topicsClient.callService(request, function(result) {
        console.log("Getting topics...", result);
        var resultsSorted = result.topics.sort();
        for(var i=0; i < resultsSorted.length; i++) {
            // Up until here, I successfully retrieve the topic names
            // Now I want to get the topic type for each topic name, this is not working
            var topicTypeClient = new ROSLIB.Service({
              ros : ros,
              name : '/rosapi/topic_type',
              serviceType : 'rosapi/Topics'
            });

            var request = new ROSLIB.ServiceRequest();
            topicTypeClient.callService(request, function(result) { 
              console.log('Topic Type: ', result.name);  // Guessing here as to how I would access the topic 
            });
        }
      });
    };

How do I go about getting the topic types? Any suggestions would be greatly appreciated!

Thanks
Racehl

Behery, 2018-06-21:
Hi Rachel, 
there's a service in the rosbridge_suite called TopicType that does this. It's called `/rosapi/topic_type` and is defined [here](rosbridge_suite/rosapi/srv/TopicType.srv). You can call it from `roslibjs` as you did above for listing the topics.

rachelspykerman, 2018-06-21:
So I tried the following, and it doesn't work. Any suggestions? Shouldn't I have to specify the topic name somewhere?

      var topicsClient = new ROSLIB.Service({
        ros : ros,
        name : '/rosapi/topics',
        serviceType : 'rosapi/Topics'
      });

      var request = new ROSLIB.ServiceRequest();
      topicsClient.callService(request, function(result) {
        console.log("Getting topics...", result);
        var resultsSorted = result.topics.sort();
        for(var i=0; i < resultsSorted.length; i++) {
            //console.log(resultsSorted[i]);
            var topicTypeClient = new ROSLIB.Service({
              ros : ros,
              name : '/rosapi/topic_type',
              serviceType : 'rosapi/TopicType'
            });

            var request1 = new ROSLIB.ServiceRequest();
            topicTypeClient.callService(request, function(result) {
              console.log('Topic Type: ', result.name);
            });
        }
      });
    };

Behery, 2018-06-21:
You specify the topic name as a parameter in the service request as follows:
```
var request = new ROSLIB.ServiceRequest({topic: 'name_of_topic'});
```

The definition for the service is here: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosapi/srv/TopicType.srv (it was incorrect in my last comment, sorry)

There's also a tutorial here: http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality

rachelspykerman, 2018-06-22:
Sorry for the delay. I'm still running into issues with the following code:

    function getTopics() {
      var topicsClient = new ROSLIB.Service({
        ros : ros,
        name : '/rosapi/topics',
        serviceType : 'rosapi/Topics'
      });

      var request = new ROSLIB.ServiceRequest();
      topicsClient.callService(request, function(result) {
        console.log("Getting topics...", result);
        var resultsSorted = result.topics.sort();
        for(var i=0; i < resultsSorted.length; i++) {
            //console.log(resultsSorted[i]);
            var topicTypeClient = new ROSLIB.Service({
              ros : ros,
              name : '/rosapi/topic_type',
              serviceType : 'rosapi/TopicType',
              topic : resultsSorted[i]
            });

            var request1 = new ROSLIB.ServiceRequest();
            topicTypeClient.callService(request, function(result) {
              console.log('Topic Type: ', result.topic_type);
            });
        }
      });
    };

The basic ros functionality tutorial doesn't go over requesting topic types, only topic names (which I have working). The result.topic_type ends up being undefined (also tried result.name). Am I referencing the wrong parameter to access the topic type? 

Rachel

Behery, 2018-06-22:
it's a normal service that you can call. Add the topic name to the service request (like in my previous comment)
```
var topicTypeClient = new ROSLIB.Service({
          ros : ros,
          name : '/rosapi/topic_type',
          serviceType : 'rosapi/TopicType',
        });

        var request1 = new ROSLIB.ServiceRequest(
        {topic : resultsSorted[i]});
        topicTypeClient.callService(request, function(result) {
          console.log('Topic Type: ', result.type);
        });
```

and i just noticed that the `topics` service also returns a list of types. so you don't need to call the second service.

you can do this:
```
var topicTypeClient = new ROSLIB.Service({
          ros : ros,
          name : '/rosapi/topics',
          serviceType : 'rosapi/Topics',
        });
topicTypeClient.callService(request, function(result) {
          console.log('Topics: ', result.topics); //topic names
console.log('Topic Types: ', result.types); //topic types 
        });
```
the later service is defined here : https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosapi/srv/Topics.srv

rachelspykerman, 2018-06-22:
Great thank you! I got it working now :)