RobotWebTools/roslibjs #297: The version of the socket.io is too low,Can you upgrade it?

helpceo, 2018-07-31:
Dependency:
    "socket.io": "1.4.8"----"socket.io": "2.1.1"
    "uws" is abolished

chrisl8, 2019-01-18:
There is actually a security vulnerability attached to a dependency of socket.io 1.4.8

I am getting flagged for security issues just for using roslibjs as a dependency now. :-1:

chrisl8, 2021-04-13:
socket.io is causing a security audit failure again, along with xmldom.

MatthijsBurgh, 2021-04-13:
The dependencies have been update on the develop branch. These changes will be released in the near future, probably in a month.

P.S I would consider it bad practice to respond on such old stale issues. Please open a new issue in the future.

closed 2021-04-13 by MatthijsBurgh

chrisl8, 2021-04-13:
I agree, but I didn't want to open a new issue when I was 90% sure it was already being worked on and this issue was still open when I searched for socket.io.

Thank you, I appreciate the work being done here.