RobotWebTools/roslibjs #302: Websocket.js: ReferenceError: global is not defined

JulianAssmann, 2018-10-26:
When I try to use roslibjs in angular and the @types/roslib package, i get the following error: 

```
ReferenceError: global is not defined
./node_modules/roslib/src/util/shim/WebSocket.js
WebSocket.js:1
__webpack_require__
bootstrap:78
./node_modules/roslib/src/core/Ros.js
Ros.js:6:16
__webpack_require__
bootstrap:78
./node_modules/roslib/src/core/index.js
index.js:4:9
__webpack_require__
bootstrap:78
./node_modules/roslib/src/RosLib.js
RosLib.js:18:15
__webpack_require__
bootstrap:78
./src/app/shared/ros.service.ts
http://localhost:4200/main.js:263:64
__webpack_require__
bootstrap:78
./src/app/connection/connection.component.ts
http://localhost:4200/main.js:216:77
__webpack_require__
bootstrap:78
./src/app/app.module.ts
http://localhost:4200/main.js:158:90
__webpack_require__
bootstrap:78
./src/main.ts
main.ts:1
__webpack_require__
bootstrap:78
0
http://localhost:4200/main.js:378:18
__webpack_require__
bootstrap:78
checkDeferredModules
bootstrap:45
webpackJsonpCallback
bootstrap:32
<anonymous>
http://localhost:4200/main.js:1:2
```

JulianAssmann, 2018-10-27:
I saw #286 will solve this problem. It only needs to be merged.

viktorku, 2018-11-06:
The PR has been merged. I'm gonna go ahead and close this.

closed 2018-11-06 by viktorku