RobotWebTools/roslibjs #313: RWT Instantly Closing Connection

CycleMark, 2019-01-15:
asked 39 hours ago

MarkyMark2012 gravatar image
MarkyMark2012 
1592 ●78 ●95 ●118
Hi,

Is anyone using RWT (ROS WebTools?). I've set up both rosbridge and the client roslibjs client side libs etc.

When trying to run even the simple.html from

https://github.com/RobotWebTools/roslibjs/blob/develop/examples/simple.html

The connect is opened on the rosbridge side and instantly closed again

e.g.

[INFO] [1547420728.703490]: Client connected.  1 clients total.
[INFO] [1547420728.709122]: Client disconnected. 0 clients total.

Is anyone is successfully using RWT I'd be really interested to hear.

Thanks

Mark

mvollrath, 2019-01-19:
Works for me on develop vs. rosbridge 0.10.1.  I followed the instructions.

CycleMark, 2019-01-20:
Hi,

Is RWT deprecated then? 

As the current version of rosbridge is version 2.0 http://wiki.ros.org/rosbridge_suite

Thanks
Mark

jihoonl, 2019-01-21:
RWT isn't deprecated. The version 2.0 means it is using [rosbridge protocol 2.0](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md)

CycleMark, 2019-01-21:
Hi,

Pleased to hear RWT isn't deprecated :)

I've reinstalled as per el al

http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge

But the connect is still disconnect immediately - I've attached one of the scripts I'm running (modified because the web locations do not appear to exist)

As per the simple.html it's still closing the connection
[ROS.zip](https://github.com/RobotWebTools/roslibjs/files/2780238/ROS.zip)

Advise much appreciated

Thanks
Mark 

Output

 roslaunch rosbridge_server rosbridge_websocket.launch --screen... logging to /home/mark/.ros/log/e08ab316-1dae-11e9-a9a6-f7a6c072517a/roslaunch-kobukiMxnet2-18070.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.79:33044/

SUMMARY
========

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address:
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosbridge_websocket/use_compression: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

auto-starting new master
process[master]: started with pid [18080]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e08ab316-1dae-11e9-a9a6-f7a6c072517a
process[rosout-1]: started with pid [18105]
started core service [/rosout]
process[rosbridge_websocket-2]: started with pid [18122]
process[rosapi-3]: started with pid [18123]
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [1548097259.828494]: Rosapi started
[INFO] [1548097260.691081]: Rosbridge WebSocket server started on port 9090
[INFO] [1548097266.138415]: Client connected.  1 clients total.
[INFO] [1548097266.146736]: Client disconnected. 0 clients total.
[INFO] [1548097604.213573]: Client connected.  1 clients total.
[INFO] [1548097604.220042]: Client disconnected. 0 clients total.

ubuntuslave, 2019-01-29:
I'm getting the same behavior as @CycleMark . Did you find out the problem or how to overcome it? I get the instant disconnection on either Firefox and Chrome browsers.

CycleMark, 2019-01-29:
@ubuntuslave - Just FYI I've not found a solution yet. I did find someone on another forum that seemed to get it to work but with a really old browser. I don't really want to go down that route

mvollrath, 2019-01-30:
I would try making sure all of the important libs are coming from the right place.  If any of the Python libs used by rosbridge (especially Tornado) are installed on your system via pip it will infect your environment.  If you import a module and its `__file__` is any other root path than `/usr/lib/` try removing it from global `pip`.  You might also check for rosbridge libs installed in weird places.

Also confirm package versions with `dpkg` on a Debian system:
```
$ dpkg -l | grep rosbridge
ii  ros-kinetic-rosbridge-library                               0.10.1-0xenial-20181217-091951-0800                   amd64        The core rosbridge package, repsonsible for interpreting JSON andperforming the appropriate ROS action, like subscribe, publish, call service, and interact with params.
ii  ros-kinetic-rosbridge-server                                0.10.1-0xenial-20181217-092826-0800                   amd64        A WebSocket interface to rosbridge.
ii  ros-kinetic-rosbridge-suite                                 0.10.1-0xenial-20181217-093131-0800                   amd64        Rosbridge provides a JSON API to ROS functionality for non-ROS programs.
```

CycleMark, 2019-01-30:
Hi,

Thanks for the reply. I've tried both uninstalling and reinstalling rosbridge via  

sudo apt-get remove ros-kinetic-rosbridge-*
sudo apt-get install ros-kinetic-rosbridge-*

As well as downloading the git repo, building that locally and sourcing it and running from there (after apt-get removing just to be sure)

Also when installed as a package the versions match yours above.

In all cases I am still getting the same issue - client is connecting then immediately disconnecting for no apparent reason. Using --screen doesn't appear to yield any more information as to why it's failing.

Is there something else I can try?

Kind Regards

Mark

jihoonl, 2019-01-30:
I think you are using cdn version of roslisbjs which needs updates. Can you try to use roslib.js in develop branch in your example page?

CycleMark, 2019-01-30:
Hi - Just doubled checked and still the same thing. I've just downloaded the dev branch of:

Client Side
roslibjs
ros2djs

Server Side
rosbridge_suite

And recompiled (Server Side) still no luck. 

Below is my whole client side repo including necessary libraries - it looks okay to me (Using ros2djstext.html specifically)

[ROSRWT.zip](https://github.com/RobotWebTools/roslibjs/files/2812354/ROSRWT.zip)

Thanks

Mark

ubuntuslave, 2019-02-04:
I was able to get the `examples/simple.html` on a different machine. Thus, that got me thinking why just my machine was working. I was able to pinpoint the problem to my **Python** environment. I was using `pyenv` to set my python environment to 2.7.15.  This is what I did to get it working:

1. Uninstall *tornado* and *pymongo* using `pip`.
2. Uninstall *rosbridge_suite* with `$ sudo apt remove ros-kinetic-rosbridge-suite`
3. Switch back to my `$ pyenv global system` version of python. 
4. Reinstall *rosbridge_suite* with `$ sudo apt install ros-kinetic-rosbridge-suite`
5. Run the server with `roslaunch rosbridge_server rosbridge_websocket.launch`
6. Go into the `examples` folder and run some example in the browser (I used Firefox). Follow the indicated commands for the particular example. 

I can now see the ROS messages flowing in the browser's console.

CycleMark, 2019-02-13:
@ubuntuslave - many thanks!! Got it little *&*^ to work. In my case all I had to do was:

$ sudo apt remove ros-kinetic-rosbridge-suite
$ sudo apt-get remove python-tornado
$ sudo apt install ros-kinetic-rosbridge-suite

It's now firing up and making and keeping the connection.  Some kind of conflict with tornado installs is my guess.

Closing issue.

Thanks
Mark

closed 2019-02-13 by CycleMark

dogoepp, 2019-02-14:
I had the same issues with ros melodic, and had to remove a globally installed tornado (through pip). This issue deserves a little documentation in the installation instructions of rosbridge. It is indeed not a common issue for me, and I would not be able to tell that it is tornado's version that prevents rosbridge from working.

mvollrath, 2019-02-14:
I think the biggest issue is that Tornado handlers are failing silently, making it really difficult to detect and troubleshoot Tornado-related problems.  Exception logging for Tornado handlers would be a welcome improvement.

dogoepp, 2019-02-14:
I assume that you are quite right. However, is there a chance that this true issue be solved anytime soon ? Should it not be the case, the above discussion is quite relevant for troubleshooting.

mvollrath, 2019-02-14:
https://github.com/RobotWebTools/rosbridge_suite/pull/386

kennedywai, 2019-02-25:
> @ubuntuslave - many thanks!! Got it little _&_^ to work. In my case all I had to do was:
>
> $ sudo apt remove ros-kinetic-rosbridge-suite
> $ sudo apt-get remove python-tornado
> $ sudo apt install [...]

Hi, I am having the same problem and issues with you. I followed your commands and it still didnt work, anything else that I can try?

kennedywai, 2019-02-25:
Issue solved. The lastest release is not working for me on Nvidia Jetson TX2. But it works on my laptop. I downgraded to https://github.com/RobotWebTools/rosbridge_suite/releases/tag/0.8.6 and then it works now...

orzechow, 2019-03-14:
Thanks guys, I had the same issue.
Turned out I hav a tornado installed in pip, via apt only ros-kinetic-rosbridge-server, not the whole suite, etc.

Uninstalling all and then installing the suite with 
```bash
$ sudo apt install ros-kinetic-rosbridge-suite
```
fixed the problem.

Maybe roslibjs should have a documentation about proper setup?

RDaneelOlivav, 2019-10-16:
Great, thanks for the solution of the issue.
For me it fixed it uninstalling everything and reinstalling bridge:

> sudo apt update
> sudo apt remove ros-kinetic-rosbridge-suite
> sudo apt-get remove python-tornado
> sudo pip uninstall tornado
> sudo apt install ros-kinetic-rosbridge-suite

It would surely help people knowing this, or just fix the issue transparently for the user.

Rezenders, 2021-04-23:
I am having a similar issue but only when calling the function that connects to ROS using a button in my html.

The proposed solution did not work:
> sudo apt update
> sudo apt remove ros-kinetic-rosbridge-suite
> sudo apt-get remove python-tornado
> sudo pip uninstall tornado
> sudo apt install ros-kinetic-rosbridge-suite

I have a button:
`<button id="robot_ip_submit" type="submit" form="robot_ip_form" value="Submit" onclick="connectRobotIP()">Conect</button>`

And the connectRobotIP() function is:
```
function connectRobotIP(){
    var robot_ip = document.getElementById("ip_robo").value;
    ros.connect('ws://'+String(robot_ip)+':9090');
}
```

If I call this same function from the console or from the body of the javascript code it works. Any ideias? I am using flask btw.