RobotWebTools/roslibjs #315: cbor compression makes an error : "Uncaught Cannot process BSON encoded message without BSON header."

HeekonKim, 2019-01-22:
I wrote a simple code like below and included it in an html file.

`
<script type="text/javascript" src="http://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
<script type="text/javascript" src="http://static.robotwebtools.org/roslibjs/current/roslib.min.js"></script>
<script src="bower_components/cbor/cbor.js" type="text/javascript"></script>
<script>
  var ros = new ROSLIB.Ros({
    url : 'ws://localhost:9090'
  });

  ros.on('connection', function() {
    console.log('Connected to websocket server.');
  });

  ros.on('error', function(error) {
    console.log('Error connecting to websocket server: ', error);
  });

  ros.on('close', function() {
    console.log('Connection to websocket server closed.');
  });

  var listener = new ROSLIB.Topic({
    ros : ros,
    name : '/os1_node/points',
    messageType : 'sensor_msgs/PointCloud2',
    compression : 'cbor',
  });

  console.log(listener);
  listener.subscribe(
    function(message) {
  } 
);
</script>
`

I did `'roscore',` `'roslaunch rosbridge_server rosbridge_websocket.launch'`, and played an appropriate `rosbag` file, which has a topic named `'/os1_node/points'`.

But the browser console makes an error "Uncaught Cannot process BSON encoded message without BSON header." at roslib.min.js:1.

The cbor-compressed message can't be decoded because the error comes out even before the function called in listener.subscribe() works.

mvollrath, 2019-01-22:
This is happening because the CDN version of roslibjs has not been updated with CBOR support.  You can try building roslibjs from the develop branch and loading that instead, or wait for the next release which is [currently in progress](https://github.com/RobotWebTools/roslibjs/pull/314).

jihoonl, 2019-01-23:
will update cdn within this week.

HeekonKim, 2019-01-23:
Thank you for answer. I built from the develop branch and did the same thing but same error occurs. Is there a code using cbor compression that I can refer to?

mvollrath, 2019-01-23:
[roslib_bench](https://github.com/EndPointCorp/roslib_bench) has a docker example.

EzraBrooks, 2025-11-09:
This looks like it may have been fixed years ago :)

closed 2025-11-09 by EzraBrooks