RobotWebTools/roslibjs #321: removeListener only takes instances of Function

baptistemanson, 2019-02-22:
From time to time, we get those logs in prod:
```
api | /usr/src/api/node_modules/eventemitter2/lib/eventemitter2.js:541
api | throw new Error('removeListener only takes instances of Function');
api | ^
api |
api | Error: removeListener only takes instances of Function
api | at RosTCP.EventEmitter.off (/usr/src/api/node_modules/eventemitter2/lib/eventemitter2.js:541:13)
api | at Topic.unsubscribe (/usr/src/api/node_modules/roslib/src/core/Topic.js:128:14)
```

We call unsubscribe like this:
```js
topic.unsubscribe()
```
It seems to be a race condition.

biw, 2019-02-25:
This might be fixed by upgrading to [`1.0.1`](https://github.com/RobotWebTools/roslibjs/releases/tag/1.0.1)

biw, 2019-02-25:
This is due to calling `topic.unsubscribe()` before `topic.subscribe()`. Created a fix in https://github.com/RobotWebTools/roslibjs/pull/324

baptistemanson, 2019-03-07:
Your patch fixed the issue thanks!

closed 2019-03-07 by baptistemanson