RobotWebTools/roslibjs #322: What server to run for socket.io support

Rayman, 2019-02-25:
I'm trying to connect to rosbridge via socket.io as explained in the docs:
```js
new Ros( {..., transportLibrary: 'socket.io'} );
```
What server supports this? I'm now running `roslaunch rosbridge_server rosbridge_websocket.launch` but this only supports websockets I think (I'm getting CORS errors).

moth-man, 2020-01-27:
Did you ever figure out the problem? Getting the same error

Rayman, 2020-01-28:
Unfortunately no, I'm now running a normal websocket with some reconnect logic.

VadimCiv, 2021-07-24:
Same problem here. Trying to use Ros with  transportLibrary: 'socket.io' option  but still get error "ReferenceError: io is not defined"
Please explain what I doing wrong?

package.json "socket.io": "^4.1.3",

```
var ROSLIB = require('roslib');
var io = require('socket.io');

var ros = new ROSLIB.Ros({
  transportLibrary: 'socket.io',
});

export const connectToRos = (ip) => {
  ros.connect(`ws://${ip}:9090`); // *** error here ***
};

```
Ros.js
```

Ros.prototype.connect = function(url) {
  if (this.transportLibrary === 'socket.io') {
    this.socket = assign(io(url, {'force new connection': true}), socketAdapter(this));
...
}
```

ReferenceError: io is not defined
    at Ros.push../node_modules/roslib/src/core/Ros.js.Ros.connect (Ros.js:66)

VadimCiv, 2021-08-16:
Same here ReferenceError: io is not defined
Anybody can explane how use transportLibrary: 'socket.io'?

rohanshukla9, 2022-02-24:
Even I am getting the same error, any workarounds?

EzraBrooks, 2025-11-03:
socket.io support has been removed in 2.x to decrease maintenance surface.

closed 2025-11-03 by EzraBrooks