RobotWebTools/roslibjs #327: How to install roslibjs in Ubuntu

alexlt97, 2019-03-12:
Hello I am completely new in Ubuntu thus I dont know how to install roslibjs in order to use in ROS
I tried to follow the instructions in README.md and then I download npm, sudo apt-get install npm.

So following the instruction in README.md I should follow contributions right? so I have to install node.js, that i dont exactly how I should do and then whatelse?

I am not it seems so silly but I bit lost using Ubunti and GitHub

Thanks.

Rayman, 2019-03-13:
You can skip the installation of npm if you just include the pre-build file. To do that, include the following in your html file:
```html
<script src="https://static.robotwebtools.org/roslibjs/current/roslib.min.js"></script>
```
If everyting works, you now have the ROSLIB variable in scope with all the functions.

helpceo, 2019-03-22:
你要会javascript

closed 2020-07-21 by alexlt97