RobotWebTools/roslibjs #330: roslibjs not working in melodic

dheera, 2019-04-03:
The latest version of roslibjs seems to not work in Melodic. The websocket connects fine, but getTopics(cb) doesn't call the callback. Is this issue known and is there a workaround?

dheera, 2019-04-04:
Found the solution: Downgrade tornado to 4.5.3: `sudo pip install tornado==4.5.3`

closed 2019-04-04 by dheera

ericq, 2019-09-23:
the above Downgrade tornado works. Thanks. Really want to know how it has this kind of issue.

Rayman, 2019-09-23:
The solution: don't install packages with pip. Try to only use the packages provided by `apt-get`. These have been tested (and are often a little older).

mvollrath, 2019-09-23:
Soon rosbridge will ship with Autobahn web socket server instead of Tornado, which should suffer much less from API churn.  Otherwise, as Rayman says, if you mix pip packages with ROS packages you're asking for trouble.