RobotWebTools/roslibjs #331: Service Call Response API Question

Troski, 2019-04-16:
I am currently getting familiarized with the roslibjs api, and have a question regarding ServiceCalls. How can I access the response of the service call outside of the response function? 

I tried using **Promises** but it still returns as undefined. 

  ```javascript
  wifiNetworksCB()
  {
        console.log("Networks Callback");
        var network = this.getWifiNetworks();
        console.log(network); //returns promised undefined. 
  }

    getWifiNetworks()
    {
        console.log("Getting Networks")
        var request = new ROSLIB.ServiceRequest({});
        var data;
        this.getWiFiListClient.callService(request, function getNetworks(response)
        {
            data = response.ssid[0];
            return new Promise(function(resolve, reject) {
                setTimeout(function() {
                    resolve(data);
                }, 0);
            });
        });
        return new Promise(function(resolve, reject) {
            setTimeout(function() {
                resolve(data);
            }, 0);
        });
        
    } 

```
I tried implementing some of the solutions from [Stack Overflow](https://stackoverflow.com/questions/23667086/why-is-my-variable-unaltered-after-i-modify-it-inside-of-a-function-asynchron) but with no success. 

I am sure this question is more related to javascript but I am not able to find a concrete example that enables me to you this API effectively. 

Thanks in advance!

Rayman, 2019-04-17:
Something like this? (I've modified a snipped of my own, so I don't know if this works)
```js
function getWifiNetworks() {
  return new Promise((resolve, reject) => {
    console.log("getWifiNetworks request:");
    var request = new ROSLIB.ServiceRequest({});

    this.getWiFiListClient.callService(
      request,
      response => {
        var data = response.ssid[0];
        console.log("getWiFiListClient response:", data);
        resolve(data);
      },
      err => {
        console.error("getWiFiListClient err:", err);
        reject(err);
      }
    );
  });
}
```

Troski, 2019-04-17:
Thanks @Rayman! this works! great!

Troski, 2019-04-17:
The solution from @Rayman  solves my question perfectly.

closed 2019-04-17 by Troski

chan1914, 2020-08-19:
I have the same problem with roslibjs database call, and have a question regarding ServiceCalls. I have no idea how to solve the problem even with Promises in Javascript.
I am currently working in Express and node.js. How can I access the response of the service call outside of the response function?

I tried using Promises but it still returns as undefined.
[image]

Rayman, 2020-08-19:
1. `ros` and the service should probably a member to keep the connection open
2. `getTorquelog` is a function which returns a promise to the service call result. That's where you access the value

P.s. This is not really a forum to ask questions, it's a bug tracker. Questions are more suited in https://answers.ros.org/