RobotWebTools/roslibjs #332: Multiple MessageType within a topic.

christopherholmes, 2019-05-09:
Hi, 

I have two ROS (.msg) files, (Point_Array.msg) and (Point.msg). 

Point_Array message:
parray = []

Point message
float32 x
float32 y
float32 z

I want to populate the Point_Array message with the Point_message, the Point_Array needs to hold 10 points. 

How do I build the message in my js file. 

My current function looks like this:

```javascript
this.Path = new ROSLIB Topic({
   ros : this.ros,
   name : '/Path'
   messageType : 'my_msgs/Point_Array'
});

var final_points = new ROSLIB.Message({
   parray : {
       Point : {
       }
   }
})

var tempArray = []

for (var i = 1; i < this.my_data.length-1; i++){
   var points = new ROSLIB.Message({
   Point : {
          x : parseFloat(my_data[i].X)
          y : parseFloat(my_data[i].Y)
          z : parseFloat(my_data[i].Z)
   }   
})

tempArray.push(points)
}

final_points = new ROSLIB.Message(tempArray)
this.Path.publish(this.final_points)
```

Can anyone help with this at all?

Rayman, 2019-05-09:
I think that the construction of `final_points` is wrong. Can you try this?
```js
var final_points = new ROSLIB.Message({
  parray: tempArray
});
this.Path.publish(this.final_points)
```

christopherholmes, 2019-05-09:
Hi Rayman,

Thanks for your help.

I can push the the roslib message of points into a normal java array set by set in a for loop - and after loop output array to console it all looks correct.
However I cannot add the same roslib message points to the roslib message array (basically message in message) as the data type of the array is not the parray type.

Do you know how to run a loop and append another rosmsg type into that message array?