RobotWebTools/roslibjs #334: Do you plan to add QoS as optional parameter when creating a subscription / publication ?

adrienbarral, 2019-05-26:
I understand that this library must remain compatible with ROS 1, but I guess that it will be possible to add QoS as an optional parameter when we instanciate a new Subscriber or Publisher.

The problem is that today, roslibjs doesn't allow to control QoS, so some connections doesn't work, because QoS in C++ doesn't match with QoS in JS. As an example I have an unreliable data published in my robot, and I can't subscribe to it in my web interface. This because in ros2 webridge, subscriptions are created with default QoS, which doesn't match the QoS for unreliable datas.

So my questions are : 
* Do you plan to release a roslibjs for ros 2 ?
* Do you plan to modify roslibjs in a near future to take QoS into account ?
* If I do modifications in ros 2 web bridge and in roslibjs to allow controlling QoS of data,  Will you accept these modifications ? (if the code quality check is OK, of course)...

If you are interested in these modifications, I can describe to you here how I plan to do it.

Rayman, 2019-05-28:
(I'm not a maintainer, but) if you are planning to make changes to the protocol, you could create a PR for this file. Then everybody can have a discussion around that.
https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md

mikelyndersOKCC, 2020-02-27:
@adrienbarral any developments on this?

adrienbarral, 2020-03-02:
@mikelyndersOKCC :  No I had no time to work on it. What I did is that I create my front end application in QT. I embed my webpage in the QT WebView, and I created a ros2 C++ module that I use as a proxy between ROS2 and the webpage (thanks to the QT Webchannel which is either a WebSocket or a Chrome IPC).

closed 2020-03-02 by adrienbarral