RobotWebTools/roslibjs #338: Allow compression for service calls/responses

uahic, 2019-08-12:
Hi @ all,

currently I am implementing a service-like interface for a web-app with ROS and transporting large images (gridmaps and such) is one feature. While it is supported on topics it doesnt seem to be possible to pass compression flags via services. One of my colleagues told me that protocol itself does support this for services as well and as far as I can see in the code services are also just using the same websocket-adapter implementation underneath. 

Is there any chance to get this support implemented quickly? or if I do it on my own, anything special to consider? Thanks!

mvollrath, 2019-08-12:
Rosbridge has a websocket-level compression feature (pre-message deflate) which is more efficient than protocol-level compression, for browsers which implement the standard.  Set the rosbridge_websocket.py param `use_compression` to `true`.

This will give you some fast compression, but still requires base64 packing the binary data into JSON.  What you really want is the option to use CBOR encoding for service responses.  CBOR is more efficient for binary data, but tends to be less efficient for small messages.

On the rosbridge side, there is a [TODO for service call compression](https://github.com/RobotWebTools/rosbridge_suite/blob/4b98dc71a396c5c2074005c25036647702902366/rosbridge_library/src/rosbridge_library/capabilities/call_service.py#L101).  [Topic subscriptions handle compression here](https://github.com/RobotWebTools/rosbridge_suite/blob/bd6b193d8ab917bfafc468492a7772f58a99d297/rosbridge_library/src/rosbridge_library/capabilities/subscribe.py#L318-L327).

On the roslib side, you would probably [add a compression option to the service call](https://github.com/RobotWebTools/roslibjs/blob/47336faa701ff1d7c3cda3f8a824ca6db62feb48/src/core/Service.js#L58-L64) and be done.

uahic, 2019-08-12:
@mvollrath thanks! I've implemented the roslib side (no untested :)) )

Basically I have very large images in arraybuffer (uint8) format. Easily goes up to 16MB and more. Right now I hope there is no need for tiling the images (alternative representations of the map than occupancy gridmaps) and occupancy grids itself.

uahic, 2019-08-12:
@mvollrath Pullrequest has been sent, I will have a look later on the rosbridge side but I do lack total knowledge of all internals. Some help here would be nice! thanks in advance