RobotWebTools/roslibjs #341: Can this library used with React Native?

senakvkc, 2019-08-28:
Hey there,

I've seen this library being used with React, but I've never encountered someone using roslibjs with React Native. So my question is, can I use it in my React Native project? I'm so very new in developing applications with React Native, so I really hope this is not a very simple question. 

Thanks in advance

Rayman, 2019-08-29:
According to [this page](https://facebook.github.io/react-native/docs/network#websocket-support) websockets are supported in react native, but I've never tried it. Let us know if it works :)

senakvkc, 2019-09-03:
Hey again, thanks for the info. Apparently, I can use it and works pretty much like how I wish it to. Thanks for the guidance :)

closed 2019-09-03 by senakvkc