RobotWebTools/roslibjs #342: Eventemitter2 undefined with requirejs

srottschaefer, 2019-09-19:
Hi,

I get this error in Firefox when using roslibjs with requirejs:
```
TypeError: EventEmitter2 is undefined
TypeError: roslib.Ros is not a constructor
```

How to recreate:
1. `git clone git@github.com:srottschaefer/bug_roslib_requirejs.git`
2. npm install
3. npx grunt
4. npm start
5. (if browser did not open automatically): open browser, go to localhost:8080 (or whatever it says in the console
6. Open developer tools console and see the error for yourself

I don't have a pull request because I am not that familiar with Javascript but maybe someone as a quick fix or workaround?

Rayman, 2019-09-20:
I've tried your example, but there is no `/js` folder. Its ignored. What should be inside?
```
"GET /js/lib/require.js" Error (404): "Not found"
```

srottschaefer, 2019-09-23:
> I've tried your example, but there is no `/js` folder. Its ignored. What should be inside?
> 
> ```
> "GET /js/lib/require.js" Error (404): "Not found"
> ```

It should be created when you do `npx grunt`. This works under Linux, I don't know about other OSes.
Grunt copies both the requirejs and roslibjs libraries from the `node_modules` folder into the `js/lib` folder. That folder is created automatically by Grunt.

Rayman, 2019-09-23:
OK, I've found it. The module is called `eventemitter2`. This should work:
```js
require.config({
    baseUrl: './',
    // The shim doesn't change anyting, got it from http://jsfiddle.net/43dCV/1/
    shim: {
        roslib: {
            deps: ['eventemitter2'],
            exports: 'ROSLIB'
        }
    },
    paths: {
        'roslib': 'js/lib/roslib',
        'eventemitter2': 'js/lib/eventemitter2'
    }
});

// This does not change anything either, see above comment.
require(['eventemitter2'], function (eventemitter2) {
    console.log('eventemitter2:', eventemitter2);
})

// This fails
require(['roslib'], function (roslib) {
    console.log('roslib:', roslib);
    ros = new roslib.Ros({
        url: 'ws://localhost:9090'
    });
});
```

srottschaefer, 2019-09-24:
Tested it, still the same messages. I think the problem isn't that eventemitter2 isn't loaded by the browser (it is available in the web developer console) but that the packaged roslib does some stuff when using it in combination with requirejs so that it sets the EventEmitter2 variable to undefined.

Rayman, 2019-09-24:
There is also an issue that `eventemitter2` is bundled inside `roslib.js`. I've send a PR here https://github.com/RobotWebTools/roslibjs/pull/344. This gets in the way of requirejs. 

I've uploaded your example with the patched `roslib.js` here: https://roslibjs-bug-requirejs.netlify.com. You can see it working and download the `roslib.js` from there.

Also, when you are working, make sure to refresh with "empty cache and hard reload". Chrome caches the javascript files hard.

closed 2019-09-24 by srottschaefer