RobotWebTools/roslibjs #343: Duplicate subscribe messages!

fabioaraujopt, 2019-09-23:
I'm using the following configuration: 

`var ros = new ROSLIB.Ros({
   url : 'ws://x.x.x.x:9090'
});

listener=new ROSLIB.Topic({
    ros : ros,
    name : '/vigil/sensor/gps/navsatfix',
    messageType : 'sensor_msgs/NavSatFix'
});

listener.subscribe(function(message){
	listener.unsubscribe();
	console.log(message);
});
`
I placed the subscribe function inside a javascript setInterval, in order to call it everytime i need data, instead of being always listening to broadcast. However every time a new subscribe is done. The number of response doubles. (I think it receives all messages from begin).

`first_call -> 1 log
first_call -> 3 logs 
first_call -> 4 logs 
first_call -> 5 logs 
first_call -> 6 logs `

Rayman, 2019-09-23:
Quote from the documentation

#### unsubscribe(callback)
Unregisters as a subscriber for the topic. Unsubscribing stop remove all subscribe callbacks. **To remove a call back, you must explicitly pass the callback function in.**

callback: the optional callback to unregister, if * provided and other listeners are registered the topic won't * unsubscribe, just stop emitting to the passed listener

fabioaraujopt, 2019-09-24:
Can you show me example? Didn't understand

Rayman, 2019-09-24:
Something like this (not tested)
```js
function handler(message) {
	console.log(message);
	listener.unsubscribe(handler);
}

listener.subscribe(handler);
```

asvinp, 2020-05-18:
> Can you show me example? Didn't understand

Hey, did you figure out the solution for this?

ManuelZ, 2021-02-18:
@asvinp @fabioaraujopt Any solution?

ManuelZ, 2021-02-18:
I had this issue when using React Hooks.

I was clicking a button to get subscribed and then again the same button to get unsubscribed from a topic.
- I had my topics declarations inside my component... solution, to move them out.
- I needed to  memoize the function that I was using to switch the subscription on/off and as well memoize the function callback passed to topic.unsubscribe()

asvinp, 2021-04-15:
> I had this issue when using React Hooks.
>
> I was clicking a button to get subscribed and then again the same button to get unsubscribed from a topic.
>
> * I had my topics declarations inside my [...]

If possible could you show an example?

EzraBrooks, 2025-11-04:
This looks like it came from incorrect usage in a React application, not a bug in roslibjs.

closed 2025-11-04 by EzraBrooks